andreiw/RaspberryPiPkg #1: No Response From USB Keyboard

gordan-bobic, 2018-01-04:
I am trying to get this to work on my Pi 3, and I am failing.
I built RPI_EFI.fd natively on an aarch64 machine using GCC 6.
I placed it along with bootcode.bin, fixup.dat and start.elf from the firmware repository at tag bcd3b8ff86949659e9eec20fd46bf27a2fcb971a on FAT partition 1 on the micro SD card.

config.txt:
arm_control=0x200
enable_uart=1
armstub=RPI_EFI.fd
disable_commandline_tags=1

Tianocore boots (raspberry logo comes up), but doesn't respond to keyboard input, i.e. ESC does nothing. I tried two different USB keyboards. Is there a special requirement for keyboard compatibility?

If I add startup.nsh containing the path to grubaa64.efi, it boots grub, but that, too doesn't respond to keyboard input.

Is there anything additional I have to do to get the keyboard input working?

andreiw, 2018-01-04:
USB keyboard support is pretty early and has so far been validated with a wireless (no name?) keyboard I had on hand.

It would be helpful to know the make and model of the keyboard that doesn’t work. For now I’d avoid Apple keyboards or keyboards with USB hubs, just to keep things simple.

Please attach a serial cable and get me the output of a Tiano boot with the DEBUG image.

andreiw, 2018-01-04:
Also, I had fixed some bugs Dec 30 in the periodic transfer DwUsbHostDxe patch, so be sure to refresh the tree if you checked out the earlier version (circa Dec 26th)

gordan-bobic, 2018-01-04:
I'll see if I can find another keyboard, and I'll have to build the debug image again (my build directory was on tmpfs). In the meantime, does the RPI_EFI.fd from the link below work for you?
http://ftp.redsleeve.org/pub/misc/RPI_EFI.fd

andreiw, 2018-01-04:
I won’t be able to try until the 10th. USB mass storage works fine, I assume? And you’re able to use the serial console?

andreiw, 2018-01-04:
You still haven’t gotten back to me with the make and model of the keyboards that do not work.

gordan-bobic, 2018-01-04:
I haven't tried USB storage, I am booting off the SD card.

Speaking of which, what is OTP in this quote from the README?
"UEFI boot media can be an SD card or USB mass storage, if you've enabled USB booting previously in the OTP (i.e. via program_usb_boot_mode=1)."

I'll try and get the serial console working.

The keyboards I tried so far are:
Microsoft Comfort Curve 2000
Microsoft Comfort Curve 3000
Old Microsoft keyboard with a dual USB/PS2 capability with a built in USB hub

andreiw, 2018-01-04:
OTP -> one time programmable, a set of fuses that control some of the low-level RPi3 

https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/bootflow.md

andreiw, 2018-01-04:
You can continue booting from a SD card, I just want to understand if USB sticks work for you (say from the Shell)

gordan-bobic, 2018-01-04:
I was about to use USB devices like the keyboards in question with u-boot, if that helps. I haven't tried USB storage.

gordan-bobic, 2018-01-04:
Any chance you could post a "release" tarball containing a known good combination of the firmware blobs and RPI_EFI.fd? That would save a LOT of time.

Another thing I noticed is that the timer countdown in grub is _very_ slow. It takes 2-3 seconds to tick down by 1 second.

andreiw, 2018-01-04:
That’s a great idea (and thanks for your additional bug reports). Please file these separately, so that you will be notified when they are resolved.

andreiw, 2018-01-13:
Beyond whatever no-name keyboard I have originally tested, I've verified that the Logitech K750 works great. I'll try some wired keyboards later.

Please get yourself a serial cable, my debug prebuilt (Binary/prebuilt/2018Jan13-GCC49/DEBUG) and share the UEFI boot log you get for a non-working keyboard.

andreiw, 2018-01-24:
Okay, this is getting resolved as closed.

It's expected that a) there could be some keyboard that may not work b) that keyboard support will improve, modulo the phase of the moon. Since a) I neither have the keyboard models referenced above for testing b) confirmation that the latest build does not work correctly, I'm resolving this with a documentation change to set the expectations correctly.

If you have a non-working keyboard and a debug log to share, please file another issue in the future.

andreiw, 2018-02-22:
The USB support for keyboards has been improved.

This has been now tested with:
* Dell SK-8125 keyboard (with built-in hub)
* Microsoft Natural Ergonomic Keyboard 4000

closed 2018-02-22 by andreiw

andreiw, 2018-02-22:
Fixed as of https://github.com/andreiw/RaspberryPiPkg/tree/master/Binary/prebuilt/2018Feb22-GCC49