andreiw/RaspberryPiPkg #12: Usb Keyboard Not Working On Prompt

daveb778, 2018-04-21:
So i've gotten pretty close to getting windows 10 arm to boot. But for some reason after selecting the boot device it wont no longer detect the keyboard, when the press any key to continue prompt arrives. The numlock light will turn on but no matter what button i press nothing happens. Also since it doesnt detect a keyboard input it will go back to the last screen and the system seems to lock up at that point.
[image]

daveb778, 2018-04-21:
I've also tried two different keyboards both seem to not have an effect.

andreiw, 2018-04-21:
Do me a favor and repeat your experiment with the input being done over the serial input instead of USB. I bet you’ll discover similar behavior.

Actually, please send me a serial log of your experiments produced by a debug build of the UEFI firmware. That should help immensly.

I’ve seen something similar with winload.efi - it draws the pretty “uh-oh” blue screen, but the keys to reboot or exit to firmware don’t work - almost as if the code is waiting for a debugger or if interrupts are disabled. With a debug edk2 build I saw periodic UEFI messages stop printing as well.

thchi12, 2018-04-21:
You can try applying the wim file manually,make two partitions on your sdcard or USB,one FAT and one NTFS,then use dism to apply the image and bcdboot to create boot files.

daveb778, 2018-04-21:
Sadly i don't have a serial cable to hook up to the pi, tho im not too sure how to communicate with it that way my guess is a serial to usb and putty on my pc. I've seen people posting images of the 0xc0000225 blue screen and i have no idea how they are even getting that far.

thchi12, 2018-04-21:
Juat as I said,apply the install.wim manually

andreiw, 2018-04-22:
Folks, I've updated the readme.md file with the steps required to build arm64 WinPE USB media. You should be able to follow them provided you have a Windows 10 machine (any will do, x86, amd64 or arm64).

Also, see https://github.com/andreiw/RaspberryPiPkg/tree/master/Binary/prebuilt/2018Apr21-GCC49, which should make it more obvious what is happening. See screenshot. You should be able to repeat my "success" (for lack of a better term, given that we're crashing somewhere in NT after UEFI exits).

[image]

Since the current version (Apr 21st) and latest WoA WinPE bits seem to act much more reasonably than this ticket implies, I'm closing it as resolved.

andreiw, 2018-04-22:
Guys, take a look at https://github.com/andreiw/RaspberryPiPkg/tree/master/Binary/prebuilt/2018Apr22-GCC49 and the latest notes in readme.md.

I can get WinPE booting. If it wasn't for https://twitter.com/ntauthority's helpful hint, wouldn't be anywhere close.

[image]
[image]
[image]
[image]

andreiw, 2018-04-22:
I'll see if I can make any progress on SMP.

It would be nice to figure out where the MCCI (https://twitter.com/NTAuthority/status/957886027594641409) USB driver for DWC_OTC can be found.

The rest of the RPi3 drivers are open source and could be hypothetically build with the EWDK (https://github.com/ms-iot/bsp/tree/master/drivers) - untried, but it seems like @NTAuthority built some.

thchi12, 2018-04-22:
wow That's amazing!
Is it possible to use PSCI to get SMP?WOA needs either mppp or psci to get multicore.
According to https://zhuanlan.zhihu.com/p/34572676
Windows has built in support for BCM interrupts,and what's the problem that prevents windows from starting up?

andreiw, 2018-04-22:
This firmware already exposes PSCI support (go boot Linux and see ;-)), but the ACPI tables are as-is from MS-IoT and thus still report MPPP and no PSCI in FADT.

thchi12, 2018-04-22:
In the earlier versions there were acpi tables for  psci ,I wonder whether adding only dsdt ,csrt,dbg2 could help or not.

andreiw, 2018-04-22:
@thchi12, nope, unrelated.

andreiw, 2018-04-22:
As far as RPi3 support there appears to be a check in Hal validating PIC is not a BCM PIC. That’s why you need a debugger attached to patch out the ‘bls’ following the subtraction and compare.

daveb778, 2018-04-22:
Would it be possible to use the usb driver from the windows IoT? Cause im trying to get my hands on the file for it.

thchi12, 2018-04-22:
I don't think arm32 drivers can be used on arm64@daveb778

daveb778, 2018-04-22:
@thchi12 Good point i didnt think that through.

andreiw, 2018-04-22:
You need a 64-bit driver. IoT core is 32-bit. Hypothetically, if I understood @NTAuthority correctly, the driver might have shipped as part of the NT-based 64-bit Wndows Mobile. I don’t know much about the later, in terms of which version first had AArch64 support (8.1 or 10...).

daveb778, 2018-04-22:
@andreiw I see, i guess the does beg the question how do we get hold of it.

thchi12, 2018-04-22:
It seems that rx-130 phones may have win10m arm64

daveb778, 2018-04-22:
Oh, also trying to make the pe disk but i keep running into errors. https://puu.sh/A8iYa/2f307aa6a4.png

thchi12, 2018-04-22:
Well you can try to grab uup and make an installtion media,keeping only the efi and boot.wim under sources

daveb778, 2018-04-22:
Nvm just tried a smaller sdcard that did the trick.

thchi12, 2018-04-22:
http://tieba.baidu.com/p/5650553455?share=9105&fr=share&see_lz=0&sfc=copy&client_type=2&client_version=9.4.8.4&st=1524405579&unique=17AB6B5F22E960726FBC25BC1894B03E
Here's my only known win10m aarch64 .(Don't worry as English translation is done below.I don't know whether it helps or not

andreiw, 2018-04-22:
Yeah you pretty much need the serial adapter today to have that windbg connection, until I figure out how to hot-patch the Hal... That’s probably where most of my interest around WoA,  today, after solving the PSCI issue. So if someone wants to focus on locating and validating the USB controller driver (https://docs.microsoft.com/en-us/windows-hardware/manufacture/desktop/add-and-remove-drivers-to-an-offline-windows-image), that would be fantastic.

andreiw, 2018-04-22:
@daveb778 until I add high-speed support to the SD host controller UEFI driver, I recommend you use USB keys for your OSes (Linux, FreeBSD, WoA), unless you really like waiting. USB is much faster. Your SD catd should pretty much just contain UEFI.

daveb778, 2018-04-22:
@andreiw Ahh kk, i've been mostly doing just this. But yea im basically stuck waiting for a serial adapter from amazon for now. Also im not too sure how to track down the usb driver, ive been looking with no luck so far.

nta, 2018-04-22:
The original USB driver was built for ARM32 only and therefore wouldn't work on ARMv8 Windows, I should probably open-source my hacky attempt at an UCX-based USB driver since I've not had time to work on it for a while now.

Also, yeah, I've noticed that the current SD UEFI driver is _slow_ when booting Windows from the SD card, other than that, wonderful work on the project in general, @andreiw. 🙂

andreiw, 2018-04-22:
@nta - wow, I feel your pain...having spent a few months off-and-on rewriting the crappy UEFI DwUsbHostDxe driver I found. Kudos for diving in and writing your own! My UEFI driver is wrong in principle (I still have no idea how this hardware is supposed to do periodic transfers), and CBI devices are still a mystery to me. Its frustrating that there are still no reasonable docs on the chip (the best I could find is https://www.quicklogic.com/assets/pdf/data-sheets/QL-Hi-Speed-USB-2.0-OTG-Controller-Data-Sheet.pdf). Maybe I should have just taken the BSD one as a base...

As far as patching the HAL I've been sort of eyeing dreamboot (https://github.com/quarkslab/dreamboot) and UEFI-Bootkit (https://github.com/dude719/UEFI-Bootkit).

Idea 1: boot WoA in EL1. Use a passthru "hypervisor" and stage 2 mappings to ensure patched EL1 VPN are either executable and not readable (then they contain patched code) or readable and XN (when code tries measuring itself we can feed it the original copy). This avoids the need to figure out how to disable patch guard.

Idea 2: rely on an EL3 agent in ATF (via an SMC) to manipulate VAs. Still have to disable patch guard, but because EL3 can use AT to translate EL1 VPNs to MPNs and access directly, we don't need to worry about non-writable mappings.

Of course, maybe some kind soul in Microsoft could just fix the check ;-D?

andreiw, 2018-04-23:
So the good news is that I have PSCI support, with all 4 cores online.

The bad news is that I have a crash. Lack of checked kernel is making this hard (I didn't find checked kernel/hal in either the 17125 EWDK or 17125 WDK.

Bug check is 0x6b (PROCESS1_INITIALIZATION_FAILED). https://neosmart.net/wiki/0x0000006b/ implies this has something to do with ntdll.dll.

BUGCHECK_P1: ffffffffc0000605 (NTSTATUS??? -> STATUS_IMAGE_CERT_EXPIRED???)
BUGCHECK_P2: 3
BUGCHECK_P3: 0
BUGCHECK_P4: 0

I don't really understand what booting with more than 1 CPU has to do with it, though...  And this is a WinPE system where everything is in RAM. Really - if I set the "Processor Enabled" flag to 0 for the other 3 GIC entries in Madt.aslc, the system boots. The must be some state between the the BSP and APs that is not identical, but I haven't yet sorted out what that could be. ATF handles all the dirty work of EL3 initialization (and it was good enough for Linux), and even the timer is configured as expected (19.2Mhz).

[image]
[image]
[image]

andreiw, 2018-04-23:
Hmm, actually the TARGET_TIME is conspicuous. RPi3 has no RTC, so it is being faked out.

The OSBUILD_TIMESTAMP is 2018-03-15 21:28:59.

In a "good" boot TARGET_TIME was 2018-03-16T01:35:12. In the 0x6b case, the TARGET_TIME is 2052-04-16T16:31:52. That does explain the cert failure...

thchi12, 2018-04-23:
Hmmm...
How about 17128+?Fetching uup files  then converting into an iso and get the boot.wim.
17128+ has no timebomb

nta, 2018-04-23:
Timebombs aren't related to these code integrity checks.

daveb778, 2018-04-23:
Oh wow that time difference, Would it be possible to add a RTC to the pi or just emulating one would probably be easier.

thchi12, 2018-04-23:
BTW I'm wondering how the time goes to 2052,I tried 14282 before this thread opened and it didn't show me the digital siganature problem.And the bomb for it should before 2016

thchi12, 2018-04-23:
Before 2016 July

falkor2k15, 2018-04-23:
How can I download and add a UEFI partition to my external hard drive that will allow Windows 10 on ARM ISO to boot up on a Raspberry Pi 3 B+? Has anyone managed to get KVM to work on qemu-system-aarch64 with graphics output at the same time?

andreiw, 2018-04-23:
@falkor2k15 KVM under what? Ubuntu/SLES arm64 on RPi3 or are you asking about Windows on Arm in a KVM VM?

falkor2k15, 2018-04-23:
Wow, excellent work and nice write-up! 

Wish I was more technical and less of a noob to understand... So the 64-bit Windows on ARM steps depend on first having the UEFI partition for Pi3 that you can then use to boot any ARM64 guests, including Linux? Not sure why WinPE comes into it when we can just grab a Windows 10 on ARM ISO as per Rafael's guide? I must be missing a concept there. Most importantly, how can we get the UEFI partition on the Pi3? 
The link within "As a starting point, take the latest prebuilt image directories and copy to empty boot media" is currently broken. There is mention of EFI.fd files, which I am more familiar in a Qemu context, but the RPi3 boot files are of the following type as you know - and they have DTB files too:
https://s9.postimg.cc/kt2y2e3y7/firmware.png

Rafael's guide can run Windows 10 on ARM using Qemu, but only with KVM disabled (due to a problem with emulating the VGA framebuffer and a lack of VirtIO and other graphics drivers), so it runs extremely slow. This is using any ARM64 Host with a KVM-enabled kernel, such as Gentoo built by sakaki.

andreiw, 2018-04-23:
@falkor2k15

Just to clarify, this has nothing to do with virtualization or KVM. RaspberryPiPkg is UEFI firmware for the bare metal Pi. Traditionally, Pi deployments either use no Arm firmware (Raspbian) or U-boot (openSuse or FreeBSD). RaspberryPiPkg allows booting 64-bit Arm operating systems that expect UEFI, such as FreeBSD, various flavors of Linux and (to a degree now) WoA.

Again, this specific thread documents getting Windows on Arm running natively on the Pi. No virtualization - Windows is started in EL2 (hypervisor privilege) and puts itself into EL1 (kernel privilege).

andreiw, 2018-04-23:
@falkor2k15

Copy https://github.com/andreiw/RaspberryPiPkg/tree/master/Binary/prebuilt/2018Apr22-GCC49/RELEASE to an SD card and you’re off to the races.

daveb778, 2018-04-23:
@falkor2k15 https://puu.sh/A9gWs/12b8c7eebc.png if you need an visual as to what i have.

andreiw, 2018-04-23:
@falkor2k15

RaspberryPiPkg has not been validated on the Pi3B+ and doesn’t suppport it.The distributed VC firmware is for the Pi3 only.

If you have a serial cable, I would like to see the log from running the DEBUG bits (not RELEASE). But please file an issue for the Pi3b+ so I keep track of it (not here).

falkor2k15, 2018-04-23:
OK thanks guys - I guess it's a B+ incompatibility then - as I just replicated the setup in Dave's pic. Well, if you guys can eventually get Windows 10 on ARM running in a useable state I'll happily downgrade to the Pi3 and even offer a donation (and encourage others) to support the work. To be honest though, I thought the next logical step for Andreiw, NT Authority and co. would be to get the KVM issues sorted out in Qemu ahead of this native Pi3 solution; that way, Windows 10 on ARM could boot up at near-native speeds (despite not being a bare-metal solution) on any ARM64 host/board (not just the Pi3).

andreiw, 2018-04-24:
The Pi efforts are going to be a bit of a science project, given the need to hot-patch Win10 Hal today just to boot. If you want something tangible, you could get https://store.hp.com/us/en/pdp/hp-envy-x2-12-e011nr?jumpid=cp_r12136_us/en/psg/envy_x2_qualcom/sticky-nav/buy-now (out of stock, again?) or the ASUS NovaGo whenever that comes out.

Alternatively, Windows has been demonstrated to run out of the box on some SBSA servers (scroll to 12:10 for Windows on the Ampere eMAG, https://youtu.be/RNzL9lFH_sA).

There is an ongoing UEFI port capable of running WoA on a $75 96board (Qualcomm 410c), that will be coming out soon, too: https://imbushuo.net/blog/archives/590

andreiw, 2018-04-24:
@falkor2k15, best way to support is to start working on building, packaging and validating the drivers (https://github.com/nta/dwusb, and https://github.com/ms-iot/bsp).

imbushuo, 2018-04-24:
My previous experimental with DB410c has 4 cores {Parking Protocol Version = 0, GIC address set}, PSCI compliant bit set in FADT. Four cores brought up fine without problems. BCM2838 might be a weird one, since it doesn't have a GIC controller.

Windows starts at EL1 (Only EL0/1 can be accessed on Qualcomm devices, EL2/3 are Qualcomm's land).

That Nokia RX-130 is another weird device. It has PSCI compliant and HVC required bit (the HVC required bit actually confuses me, my initial attempt on 410c set it too, then I got `INTERNAL_POWER_ERROR`. Removing that HVC bit makes everything fine), and Parking Protocol set to 1. MPPark is implemented in firmware. I am not sure if Windows uses PSCI or MPPark on RX-130 yet.

andreiw, 2018-04-24:
@imbushuo, how is the I/O looking on the 410c? Are these standard MMIO EHCI/SDHCI/... IP blocks, or do drivers need to be written to support current Win10 builds?

imbushuo, 2018-04-24:
@andreiw Most drivers can be reused from Little Kernel.

I have a copy of Qualcomm BSP, my initial attempt reused it a lot (so I am running cleanup now, make sure related stuffs are replaced with open-source drivers). 

Later attempts, including the one on retail Lumia 950 XL replaced them with drivers from Little Kernel. eMMC/UFS is standard MMIO. USB uses Synposys IP block too. Display is Qualcomm proprietary, but I can re-use FrameBuffer pointer from earlier stage bootloader.

There are two open source implementations of the USB controller driver, one from LK (slave mode), the other from U-Boot (host mode). EFIDroid project has done the ground work for us.

Drivers need to be written to support current Windows 10 builds. Device documentation and reference implementation (Linux though) is available.

imbushuo, 2018-04-24:
Correction: APQ8016 (DB410c) uses ChipIdea USB 2.0 IP Blocks. Later models w/ USB 3.0 use Synposys USB IP Blocks.

andreiw, 2018-04-24:
@daveb778 Did you enable debugging in the WinPE with bcdedit?

daveb778, 2018-04-24:
I did, uart is also enabled. maybe i have the pinout wrong?
https://cdn.discordapp.com/attachments/99288948077002752/438376344384962560/IMG_20180424_123035.jpg

andreiw, 2018-04-24:
That looks wrong, but I don’t have a picture for you now (can post in the evening). Try flipping TX and RX pins.

The basic test is attaching PuTTY at 115200, and you should be able to interact with UEFI. the DEBUG build has more spew.

daveb778, 2018-04-24:
[image]
Ok i see stuff in putty after that so im guessing the bcd edit didnt work, tho it said successfully completed. Is there any other way of editing the bcd?

andreiw, 2018-04-24:
Can you post your exact commands and the results of “bcdedit /store path/to/bcd”?

Also remember to safely eject media to flush fs and block buffers...

A

daveb778, 2018-04-24:
[image]
[image]
Ok, i did it all again to be sure. Yea this is very strange, im not too sure why this isnt working. 
[image]

andreiw, 2018-04-24:
Okay, keep PuTTY running and just boot Windows. You should see spew from the debugging stubs once Windows crashes.

A

andreiw, 2018-04-24:
And you’re using the latest bits, right?

A

daveb778, 2018-04-24:
What do you mean by bits?

daveb778, 2018-04-24:
Ok just using putty it seems to hang here. 
[image]

andreiw, 2018-04-24:
This is the Apr22nd build? Weird. What windows build?

A

[1874 more replies]