andreiw/RaspberryPiPkg #16: How to install on B+?

falkor2k15, 2018-05-13:
I tried with the May 9th build, but just get to the rainbow screen. I did see in the readme that there's no automatic detection, so is a modification required?

andreiw, 2018-05-13:
Try the May13th, in case you has stale bits for some reason... 

MD5 (RPI_EFI.fd) = 0923fda247b8e3aea9db7d2ce9995611
MD5 (bootcode.bin) = 17efaf1c1ef89289168d71cdc8194982
MD5 (config.txt) = 5ecb180ae22cd884d5ae5a7384d939bf
MD5 (fixup.dat) = ab0a486fd2da88c03eb3f72f1e38fc7c
MD5 (readme.md) = ba05961520ec3f8c5bc22def5efc23b3
MD5 (start.elf) = 7713734f5909d0109ff57c0ed825eb0f

Proof:

[image]

andreiw, 2018-05-14:
@falkor2k15 if you have issues please also see https://github.com/andreiw/RaspberryPiPkg/issues/14. The May 13th builds got updated with a display driver fix (and there's documented quirks for HDMI now in readme.md)

falkor2k15, 2018-05-14:
I just get a black screen, though the rainbow screen appeared briefly once before the black screen. I tried the May 13th builds (both woa-17125 only + normal). I also tried the following inside the config.txt:
arm_control=0x200
enable_uart=1
armstub=RPI_EFI.fd
disable_commandline_tags=1 

AND

arm_control=0x200
enable_uart=1
armstub=RPI_EFI.fd
disable_commandline_tags=1 
hdmi_force_hotplug=1
hdmi_group=1
hdmi_mode=4

Am I doing something wrong? 
[image]

Are we meant to hit certain keys on the keyboard after powering on the RPi3B+?

falkor2k15, 2018-05-14:
OK great - I got it to come up by pressing ESC of course! So what is up with the May 13 FD files? Will you be able to fix this in future releases, or the B+ will need it's own separate release?

falkor2k15, 2018-05-14:
Wow, I got Ubuntu 18.04 to boot up - nice!!! Unfortunately, it's not detecting the NIC.

falkor2k15, 2018-05-15:
OpenSUSE works quite well and has KVM enabled. I tried several weeks to get that to work on the Pi, but could never do it - UEFI certainly helps! And it detects the LAN. I just cannot figure out how to get LAN or Wifi to work on Ubuntu 18.04, so it cannot be installed from the mini.iso.

andreiw, 2018-05-15:
I’ve updated the May 13th bits last night, those should work...

andreiw, 2018-05-15:
I didn’t try Ubuntu yet on the Pi3B+

andreiw, 2018-05-15:
Great to hear SUSE works well!!!

falkor2k15, 2018-05-15:
I think I just need a step by step guide on how to inject the DTB file:
http://ports.ubuntu.com/ubuntu-ports/dists/bionic/main/installer-arm64/current/images/device-tree/bcm2837-rpi-3-b.dtb
I don't quite know what you do with this? But that will probably solve the Wifi issue? I know the B+ needs special LAN drivers that the normal Pi3 doesn't need (they both have different NIC models).

andreiw, 2018-05-15:
For WiFi there’s more pain...see the hoops I went through with ubuntu on the Pi 3.

To use your own dtb involves a few changes to your config.txt:

...
disable_commandline_tags=2
device_tree_address=0x8000
device_tree_end=0x10000
device_tree=my_fdt.dtb

Note: the address range must be [0x8000:0x10000). dtoverlay and dtparam parameters are also supported.

falkor2k15, 2018-05-15:
Yeah, I read that, but I don't understand. For example:
1) What is a DTB? Presumably it's a drivers file?
2) Does the DTB sit at the root of the boot partition or does it need to be installed via the operating system that we are attempting to install?
3) Do we rename it my_fdt.dtb or stick with the default DTB name?
4) How do we find out the address range? Or are we always sticking specifically with 0x8000:0x10000?

andreiw, 2018-05-16:
DTB (device tree blob) aka FDT (flattened device tree) is a binary representation of system hardware, firmware hand-off state and configuration parameters. It is not a driver, but it is what lets a kernel know what drivers to load for what hardware.

In your case simply download http://ports.ubuntu.com/ubuntu-ports/dists/bionic/main/installer-arm64/current/images/device-tree/bcm2837-rpi-3-b.dtb into the same directory as config.txt. You can name it anything you want (as you will plug the name in config.txt).

The adress range must be the one I document. That area of memory will be scanned by UEFI for your DTB. If no valid DTB is found, an internal one will be used. Regardless, the DTB is used as a template and is patched to ensure correct configuration is passed to OS.

andreiw, 2018-05-16:
I am saying 0x8000:0x10000 must be the numbers used in config.txt. You may not use different numbers.

LAN is enumerated by USB. A different dtb is not going to help - Ubuntu must just not have the USB nic driver. You could try a third party USB nic dongle, that could work...

closed 2018-05-18 by andreiw