andreiw/RaspberryPiPkg #18: USB 1.1 Mass Storage Device not working

driver1998, 2018-05-18:
I have an old USB 1.1 256MB USB stick floating around my house, which is a ZOLTRIX COOLDISK V3, and uses a SONiX SN11085A controller.
When I plug this stick in UEFI Shell, everything hang, I get no response from USB Keyboard as well as the serial terminal.
Everything is back to normal when I unplug this USB stick.
And of cause, this stick is in totally working condition when plugged in a normal PC, if you are interested in that.

This issue not necessary to fix IMO, since these USB 1.1 sticks are getting old and rare and useless these days. But since USB1 Devices are listed as untested in the README, I guess I should test it :-).

driver1998, 2018-05-18:
Interestingly, It can be recognized in WoA and show up as drive D, can't read it though.

andreiw, 2018-05-18:
Thanks, I’ll try to eBay one...

If you boot a DEBUG build with a UART connection, there will be tons of spew when it hangs. If you could get that to me, that would be fantastic.

A

driver1998, 2018-05-18:
Here is what I got, from plugging it to pulling it.
```
UsbEnumeratePort: port 3 state - 101, change - 01 on 363DFB18
UsbEnumeratePort: Device Connect/Disconnect Normally
UsbEnumeratePort: device at port 3 removed from root hub 363DFB18
UsbRemoveDevice: device 4 removed
UsbEnumeratePort: new device connected at port 3
UsbEnumerateNewDev: hub port 3 is reset
UsbEnumerateNewDev: device is of 0 speed
UsbEnumerateNewDev: device uses translator (1, 4)
UsbEnumerateNewDev: device is now ADDRESSED at 4
UsbEnumerateNewDev: max packet size for EP 0 is 64
UsbBuildDescTable: device has 1 configures
UsbGetOneConfig: total length is 32
UsbParseConfigDesc: config 1 has 1 interfaces
UsbParseInterfaceDesc: interface 0(setting 0) has 2 endpoints
UsbEnumerateNewDev: device 4 is now in CONFIGED state
UsbSelectConfig: config 1 selected for device 4
UsbSelectSetting: setting 0 selected for interface 0
InstallProtocolInterface: 09576E91-6D3F-11D2-8E39-00A0C969723B 3634FE98
InstallProtocolInterface: 2B2F68D6-0CD2-44CF-8E8B-BBA20B1B5B75 36379540
UsbConnectDriver: TPL before connect is 8, 36379718
UsbBotExecCommand: UsbBotGetStatus (Device Error)
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBootRequestSense: (Not Ready) CmdResult=0x1
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBootRequestSense: (Not Ready) CmdResult=0x1
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBootRequestSense: (Not Ready) CmdResult=0x1
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBootRequestSense: (Not Ready) CmdResult=0x1
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBootRequestSense: (Not Ready) CmdResult=0x1
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBootRequestSense: (Not Ready) CmdResult=0x1
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBootRequestSense: (Not Ready) CmdResult=0x1
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBootRequestSense: (Not Ready) CmdResult=0x1
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBootRequestSense: (Not Ready) CmdResult=0x1
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBootRequestSense: (Not Ready) CmdResult=0x1
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBootRequestSense: (Not Ready) CmdResult=0x1
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBootRequestSense: (Not Ready) CmdResult=0x1
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBootRequestSense: (Not Ready) CmdResult=0x1
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBootRequestSense: (Not Ready) CmdResult=0x1
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBootRequestSense: (Not Ready) CmdResult=0x1
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBotExecCommand: UsbBotSendCommand (Not Ready)
UsbBootRequestSense: (Not Ready) CmdResult=0x1
[... 34 lines cut ...]
UsbBotExecCommand: UsbBotSendCommand (Device Error)
UsbBootRequestSense: (Device Error) CmdResult=0x1
UsbBotExecCommand: UsbBotSendCommand (Device Error)
UsbBotExecCommand: UsbBotSendCommand (Device Error)
UsbBootRequestSense: (Device Error) CmdResult=0x1
UsbBotExecCommand: UsbBotSendCommand (Device Error)
UsbBotExecCommand: UsbBotSendCommand (Device Error)
UsbBootRequestSense: (Device Error) CmdResult=0x1
UsbBotExecCommand: UsbBotSendCommand (Device Error)
UsbBotExecCommand: UsbBotSendCommand (Device Error)
UsbBootRequestSense: (Device Error) CmdResult=0x1
UsbBotExecCommand: UsbBotSendCommand (Device Error)
UsbBotExecCommand: UsbBotSendCommand (Device Error)
UsbBootRequestSense: (Device Error) CmdResult=0x1
UsbBotExecCommand: UsbBotSendCommand (Device Error)
UsbBotExecCommand: UsbBotSendCommand (Device Error)
UsbBootRequestSense: (Device Error) CmdResult=0x1
UsbBootGetParams: UsbBootInquiry (Device Error)
UsbMassInitNonLun: UsbMassInitMedia (Device Error)
USBMassDriverBindingStart: UsbMassInitNonLun (Device Error)
UsbConnectDriver: TPL after connect is 8
UsbSelectConfig: failed to connect driver Not Found, ignore
```

BTW, I am using a 2GB SD card, not SDHC nor SDXC, and it works fine.

hansmbakker, 2019-03-03:
Is this issue still an issue with the new usb driver?

andreiw, 2019-03-03:
There is no new UEFI USB driver. There is a new Windows USB driver (courtesy of MCCI), but these have nothing to do with each other.

A

hansmbakker, 2019-03-03:
Sorry, then I mixed things up.

closed 2020-02-19 by andreiw