andreiw/RaspberryPiPkg #19: UEFI wont boot at all.

Matthew-Turner, 2018-05-19:
I am trying to boot WoA and the uefi wont boot at all. All that happens is the red light on the pi turns on but nothing else

falkor2k15, 2018-05-19:
*Ensure USB drive is under 1TB - need SDcard for windows
*Ensure you got a 100-300MB FAT32 partition on the SD before copying over the files from the WoA release to the root directory

andreiw, 2018-05-19:
@Matthew-Turner please describe in minute details exactly what you're doing.

Forget WoA. Let's just start with the basics. What's on your FAT-formatted SD card?

Matthew-Turner, 2018-05-19:
@andreiw I managed to get uefi to boot. Found out it was an issue with the way the Sd card was formatted. It was formatted as gpt so I re-formatted it as mbr and it worked😄 . Now when I try booting windows it says booting the kernel but then it just shows the loading circle constantly with no windows logo

andreiw, 2018-05-19:
Great to hear.

I suggest you join the fun in the other WoA thread.

closed 2018-05-19 by andreiw