andreiw/RaspberryPiPkg #20: Official Raspberry Pi 7" Touchscreen as Display

pixelomer, 2018-05-19:
Hi, I want to use my touchscreen as a monitor for both UEFI boot menu and the operating systems. Is it possible? I don't need touch support, I just need the display.

andreiw, 2018-05-19:
Did you try?

A

pixelomer, 2018-05-19:
Yes I did, I only got a blank screen.

andreiw, 2018-05-19:
If you have a UART cable, run the DEBUG variant if firmware and send me the full output. It is possible the resolution is too low.  What is the resolution supported?

pixelomer, 2018-05-21:
No, I don't have any UART cables. The touchscreen's resolution is 800x480. Is it too low?

andreiw, 2018-05-21:
I’ll see what I can do.

A

driver1998, 2018-05-22:
BTW, composite displays hangs at the rainbow screen.

andreiw, 2018-05-22:
What’s a “composite display”?

A

driver1998, 2018-05-22:
That is the composite video output built in to the 3.5mm jack.

andreiw, 2018-05-23:
Woah, I totally didn't know the new Pies could still do that! Very cool.

Still to look at the resolution issue, hold on!

andreiw, 2018-05-28:
This should be resolved as of the https://github.com/andreiw/RaspberryPiPkg/tree/master/Binary/prebuilt/2018May27-GCC49 build.

Let me know.

closed 2018-05-28 by andreiw

driver1998, 2018-05-28:
Both NTSC and PAL mode works, I don't have an official 7" touch screen handy so cannot test that.
BTW, you can get rid of the resolution problem (being smaller than the one got from the GPU) by specify "disable_overscan=1" in config.txt, this works for me.

pixelomer, 2018-05-28:
Windows 10 on ARM is displayed completely fine on the 7" screen. Obviously the touch screen doesn't work, it just works as a display,. We need a driver for windows to use the touch screen, right?

andreiw, 2018-05-28:
yep

A