andreiw/RaspberryPiPkg #21: AARCH64 small code model requires identical ELF and PE/COFF section offsets modulo 4 KB. GenFw: ERROR 3000: Invalid

zhouzhiwen2000, 2018-05-19:
I came across this during my build:
## log:
Building ... /home/zhouzhiwen/edk2/MdeModulePkg/Library/UefiHiiServicesLib/UefiHiiServicesLib.inf [AARCH64]
"GenFw" -o /home/zhouzhiwen/Build/RaspberryPiPkg-AARCH64/DEBUG_GCC5/AARCH64/RaspberryPiPkg/AcpiTables/AcpiTables/OUTPUT/./Madt.acpi -c /home/zhouzhiwen/Build/RaspberryPiPkg-AARCH64/DEBUG_GCC5/AARCH64/RaspberryPiPkg/AcpiTables/AcpiTables/OUTPUT/./Madt.dll
"aarch64-linux-gnu-gcc"   -g -fshort-wchar -fno-builtin -fno-strict-aliasing -Wall -Werror -Wno-array-bounds -ffunction-sections -fdata-sections -include AutoGen.h -fno-common -DSTRING_ARRAY_NAME=UefiShellNetwork1CommandsLibStrings -g -Os -fshort-wchar -fno-builtin -fno-strict-aliasing -Wall -Werror -Wno-array-bounds -include AutoGen.h -fno-common -mlittle-endian -fno-short-enums -fverbose-asm -funsigned-char -ffunction-sections -fdata-sections -Wno-address -fno-asynchronous-unwind-tables -fno-pic -fno-pie -ffixed-x18 -flto -Wno-unused-but-set-variable -Wno-unused-const-variable -mcmodel=small -c -o /home/zhouzhiwen/Build/RaspberryPiPkg-AARCH64/DEBUG_GCC5/AARCH64/ShellPkg/Library/UefiShellNetwork1CommandsLib/UefiShellNetwork1CommandsLib/OUTPUT/./AutoGen.obj -I/home/zhouzhiwen/edk2/ShellPkg/Library/UefiShellNetwork1CommandsLib -I/home/zhouzhiwen/Build/RaspberryPiPkg-AARCH64/DEBUG_GCC5/AARCH64/ShellPkg/Library/UefiShellNetwork1CommandsLib/UefiShellNetwork1CommandsLib/DEBUG -I/home/zhouzhiwen/edk2/MdePkg -I/home/zhouzhiwen/edk2/MdePkg/Include -I/home/zhouzhiwen/edk2/MdePkg/Include/AArch64 -I/home/zhouzhiwen/edk2/ShellPkg -I/home/zhouzhiwen/edk2/ShellPkg/Include -I/home/zhouzhiwen/edk2/MdeModulePkg -I/home/zhouzhiwen/edk2/MdeModulePkg/Include /home/zhouzhiwen/Build/RaspberryPiPkg-AARCH64/DEBUG_GCC5/AARCH64/ShellPkg/Library/UefiShellNetwork1CommandsLib/UefiShellNetwork1CommandsLib/DEBUG/AutoGen.c
GenFw: ERROR 3000: Invalid
make: *** [/home/zhouzhiwen/Build/RaspberryPiPkg-AARCH64/DEBUG_GCC5/AARCH64/RaspberryPiPkg/AcpiTables/AcpiTables/OUTPUT/Madt.acpi] Error 2
  WriteSections64(): /home/zhouzhiwen/Build/RaspberryPiPkg-AARCH64/DEBUG_GCC5/AARCH64/RaspberryPiPkg/AcpiTables/AcpiTables/OUTPUT/./Madt.dll AARCH64 small code model requires identical ELF and PE/COFF section offsets modulo 4 KB.
GenFw: ERROR 3000: Invalid
  WriteSections64(): /home/zhouzhiwen/Build/RaspberryPiPkg-AARCH64/DEBUG_GCC5/AARCH64/RaspberryPiPkg/AcpiTables/AcpiTables/OUTPUT/./Madt.dll AARCH64 small code model requires identical ELF and PE/COFF section offsets modulo 4 KB.
GNUmakefile:367: recipe for target '/home/zhouzhiwen/Build/RaspberryPiPkg-AARCH64/DEBUG_GCC5/AARCH64/RaspberryPiPkg/AcpiTables/AcpiTables/OUTPUT/Madt.acpi' failed

build.py...
 : error 7000: Failed to execute command
	make tbuild [/home/zhouzhiwen/Build/RaspberryPiPkg-AARCH64/DEBUG_GCC5/AARCH64/RaspberryPiPkg/AcpiTables/AcpiTables]

build.py...
 : error F002: Failed to build module
	/home/zhouzhiwen/RaspberryPiPkg/AcpiTables/AcpiTables.inf [AARCH64, GCC5, DEBUG]

- Failed -
Build end time: 21:46:07, May.19 2018
Build total time: 00:00:23
All patches applied.
##  **git log:**
commit 9b874d2d369a8f3237ff12c7084ddb7caae114ce (HEAD -> working)
Author: Andrei Warkentin <andrey.warkentin@gmail.com>
Date:   Mon May 14 01:10:50 2018 -0400

    BootGraphicsResourceTableDxe: properly handle SetBootLogo with NULL buffer
    
    SetBootLogo would not free the previous logo memory, which in turn
    means that a BGRT would still be installed, even if marked as not valid.
    Windows 10 actually uses the BGRT image for its boot screen even if
    not 'valid', which seems incredibly unintuitive.
    
    This new behavior means a few things:
    1) A boot to the default boot option (where SetBootLogo(NULL) is never
       called) will result in Windows displaying the OEM logo.
    2) Any other boot (via boot manager, second boot option, etc) will
       result in Windows displaying its own logo. Honestly, this is much
       better from a user experience, since it will be obvious that
       something "non-default" is happening to the system.
    
    Signed-off-by: Andrei Warkentin <andrey.warkentin@gmail.com>

commit 0520f839bd523f82652ae137a84acd96fca4ef58
Author: Andrei Warkentin <andrey.warkentin@gmail.com>
Date:   Sat Apr 21 22:57:53 2018 -0400

    BdsDxe: allow PlatformBootManagerWaitCallback to handle remaining timeout == 0
    
    - Properly finish drawing 100% bar.
    - Special platform behavior when timeout expires.
    
    Signed-off-by: Andrei Warkentin <andrey.warkentin@gmail.com>

commit 1caf37e9d2f86e9fc5fb1e40e77be128559d6c7d
Author: Andrei Warkentin <andrey.warkentin@gmail.com>
Date:   Sun Apr 1 02:58:31 2018 -0400

    BaseTools/tools_def: support ASLC files on AArch64
    
    These weren't being built correctly.
    
    Signed-off-by: Andrei Warkentin <andrey.warkentin@gmail.com>

commit b83e72f2c0179099072231a03dcf15709a99cc2e
Author: Andrei Warkentin <andrey.warkentin@gmail.com>
Date:   Thu Feb 22 18:28:32 2018 -0500

    UefiBootManagerLib: don't add duplicates in EfiBootManagerRefreshAllBootOption
    
    EfiBootManagerRefreshAllBootOption used to consider exactly
    identical boot options as being duplicates, but it's
    good enough to match on just type and device path, to account
    for boot options added by the PlatformBootManagerLib.
    
    Signed-off-by: Andrei Warkentin <andrey.warkentin@gmail.com>

commit 4fda6335285661457731b31f98cba66609f4abe8
Author: Andrey Warkentin <andrey.warkentin@gmail.com>
Date:   Thu Dec 14 23:41:53 2017 -0500

    MmcDxe: fix invalid HC_MMC_CSD_GET_DEVICESIZE computation
    
    This made a 64GB SDXC card look twice as small, and if
    a partition fills the entire card, then UEFI will
    ignore it, considering it to be beyond the card's end.
    
    Signed-off-by: Andrey Warkentin <andrey.warkentin@gmail.com>

commit 989f7a2cf0e27123fda5ca538b15832e115e0f4e
Author: cinnamon shia <cinnamon.shia@hpe.com>
Date:   Fri May 11 23:21:12 2018 +0800

    MdeModulePkg Variable: Fix the returned status in UpdateVariableStore
    
    If Fvb is a NULL, return EFI_UNSUPPORTED.
    If the remaining size is not enough, return EFI_OUT_OF_RESOURCES.
    
    Contributed-under: TianoCore Contribution Agreement 1.0
    Signed-off-by: cinnamon shia <cinnamon.shia@hpe.com>
    Signed-off-by: Ansen Huang <ansen.huang@hpe.com>
    Reviewed-by: Star Zeng <star.zeng@intel.com>

andreiw, 2018-05-19:
Don't see it. Most likely, you already had a stale Conf\tools_def.txt file around.

andreiw, 2018-05-19:
```
#
# Adjust to taste.
#
# This assumes the following directory structure:
#
# \
# \edk2
# \RaspberryPiPkg
# \gcc-linaro-5.5.0-2017.10-i686_aarch64-linux-gnu
# \build5
#

export WORKSPACE=$PWD
export PACKAGES_PATH=$WORKSPACE/edk2:$WORKSPACE/RaspberryPiPkg
export EDK_TOOLS_PATH=$WORKSPACE/edk2/BaseTools
export PATH=$PATH:$PWD/gcc-linaro-5.5.0-2017.10-i686_aarch64-linux-gnu/bin
export GCC5_AARCH64_PREFIX=aarch64-linux-gnu-

rm -rf Conf
mkdir Conf
make -C $EDK_TOOLS_PATH
echo $EDK_TOOLS_PATH
. $WORKSPACE/edk2/edksetup.sh BaseTools

build -b DEBUG -a AARCH64 -t GCC5 -p RaspberryPiPkg/RaspberryPiPkg.dsc
build -b RELEASE -a AARCH64 -t GCC5 -p RaspberryPiPkg/RaspberryPiPkg.dsc
```

andreiw, 2018-05-19:
This gets me a clean build.

zhouzhiwen2000, 2018-05-20:
Problem solved.Thanks!

closed 2018-05-20 by zhouzhiwen2000