andreiw/RaspberryPiPkg #22: Booting UEFI on Qemu?

falkor2k15, 2018-05-19:
Has anyone managed to get this to work? I am trying to find out how to get a display on the latest Qemu-System-Aarch64 and machine type Raspi3:

qemu-system-aarch64 -M raspi3 -m 1G -kernel kernel8.img

I tried adding a 100 MB FAT32 disk image with the UEFI files from May 13th, but something seems fundamentally wrong; I just get a black screen regardless of what I try to boot up in Qemu when emulating the pi3 machine type.

thchi12, 2018-05-19:
Try -bios kernel8.img

andreiw, 2018-05-19:
QEMU has UEFI support for Arm64. What would (sort of) emulating an RPi3 get you?

falkor2k15, 2018-05-19:
Just curiosity mainly: to check out the state of Pi3 virtualization with UEFI firmware, and see if we can re-produce all the peculiarities from testing WoA on a real physical Pi3, in the virtual world.

I'm waiting to hear back from the maintainer of this Qemu machine type code. I'm still getting a black screen, but I saw that they had a display when they used Qemu-System-Arm for the Pi1.

740291272, 2018-05-20:
I can use Other Uefi to boot it from QEMU VM

thchi12, 2018-05-20:
I'm also getting a black screen.
But pi2 works and can start Windows IoT(Though it seems lots of drivers don't work properly.

andreiw, 2018-05-20:
If you run the DEBUG variant of UEFI and get me a serial log, that would help figure out what isn’t working.

thchi12, 2018-05-20:
-m 1024
-M raspi3
-bios /sdcard/APQ/RPI_EFI.fd
-smp 4
-vnc :0
-serial file:/sdcard/APQ/output.txt
-global bcm2835-fb.pixo=0 
Running this on my phone. Only a black screen and there's nothing in output.txt, Weird

andreiw, 2018-05-20:
What serial is qemu modeling? the microUART or PL011?

740291272, 2018-05-20:
You Should Use a VGA device

thchi12, 2018-05-20:
I don't know exactly.
Using rpi2 emulation and MS's UEFI and it's output says UART..
To 740291272 when emulating rpis -device VGA doesn't work.

andreiw, 2018-05-20:
So PL011 then...

thchi12, 2018-05-20:
UART0
UEFI firmware (version  built at 09:28:57 on Oct 16 2015)
[2J[01;01H[=3h[2J[01;01H[2J[01;01H[=3h[2J[01;01H
Output when emulating rpi2, Firmware extracted from iot 10586.
-M raspi2
-m 512
-bios /sdcard/APQ/kernel.img
-smp 4
-vnc :0
-serial file:/sdcard/APQ/output.txt
-global bcm2835-fb.pixo=0

740291272, 2018-05-20:
-M raspi3
-cpu cortex-a57

thchi12, 2018-05-20:
I think cpu is unrelated...
BTW rpi3 use a53...

740291272, 2018-05-20:
Fine XD

thchi12, 2018-05-20:
Tried setting higher resolutions and it doen't work.

falkor2k15, 2018-05-20:
It seems the project started out using a modified version of Qemu before recently having the code passed to Qemu proper, but I don't think there's been much testing done since then:
https://www.raspberrypi.org/forums/viewtopic.php?f=72&t=195565

andreiw, 2018-05-23:
Until either the Qemu port support the microUART (or ATF+ UEFI can use PL011), I don't think there's anything to do here.

closed 2018-05-23 by andreiw