andreiw/RaspberryPiPkg #27: Use ACPI also for booting Linux

Googulator, 2018-06-03:
Currently it appears that only Windows is booted using ACPI, while Linux gets traditional DTBs, which are distro-specific.

Linux supports ACPI for device discovery on ARM64 since version 4.1. It would be handy to reuse the ACPI support developed for Windows also in Linux, possibly avoiding having to go distro-specific.

andreiw, 2018-06-03:
To a degree, this may be possible, but will be pointless if the end result will depend on someone adding ACPI support to critical RPi drivers in Linux.

A

woachk, 2018-06-19:
@andreiw why not adding ACPI support to those drivers then? it's doing it once..

andreiw, 2018-06-19:
Sure, why not? Due to PRP0001 the only real
place to pay attention may just be the MADT parsing (i.e. don’t try parsing for a GIC on a Pi)...

A

closed 2020-02-19 by andreiw