andreiw/RaspberryPiPkg #28: No Boot Windows On Raspberry Pi

LeazyShop, 2018-06-03:
Hello, excuse me in advance for the misspelling I am French. I have a Windows installed on the raspberry pie 3 but when I turn it on I have the boot (Screen 1 & 2) and if I press ENTER, ESC or F1 nothing happens after it starts on Shell and I do not know what to do? (Screen 3) Thanks in advance for help ^^

Screen)
1: [https://cdn.discordapp.com/attachments/424749892066607114/452858070159327246/IMG_4233.JPG](url)
2: [https://cdn.discordapp.com/attachments/424749892066607114/452863379242221568/IMG_4234.JPG](url)
3: [https://cdn.discordapp.com/attachments/424749892066607114/452863438255816734/IMG_4245.JPG](url)

thchi12, 2018-06-03:
Maybe your boot files are not correctly created.

feherneoh, 2018-06-04:
I see it detects a filesystem on USB, so try entering `bootaa64` in shell, it might start the Windows bootloader