andreiw/RaspberryPiPkg #29: When the system boot, USB port has no any current output

oniyakun, 2018-06-04:
I’m watched your video on YouTube and I try to install Windows10 in my raspberry pi3b+.

Yes, its boot and get into the system.

When I turn on the raspberry pi, I can see the usb port has current output. But when its start to load the system, the usb port's power is off.

No any current output, so I can't use my mouse and keyboard.

Hope I can fix it, thanks very much.

(Not good at Eng

feherneoh, 2018-06-04:
Installing the driver should make USB turn on after boot

thchi12, 2018-06-04:
Adding usb drivers will make windows detect your usb devices after starting up.
When starting up it won't have current.
Also...You need to remove all the usb devices due to buggy driver.

andreiw, 2018-06-05:
https://github.com/andreiw/RaspberryPiPkg/issues/12

closed 2018-06-05 by andreiw