andreiw/RaspberryPiPkg #36: Support hybrid GPT/MBR disks in UEFI

andreiw, 2018-06-13:
Merge https://github.com/Googulator/RaspberryPiPkg/commit/844a02a473b147e621e8d4709b6eaababd8fa2d9 in a TBD fashion

Googulator, 2018-06-15:
Looks like this won't be needed after all, see my comment in the megathread.

Googulator, 2018-06-17:
So, all that the tool needs to do is this (assuming ESP at sector 2048):
1. Copy 76 bytes from offset 0x10000B to offset 0xB
2. Increment byte at offset 0xF by 8

On Linux, this is trivial. On Windows, it's a PITA due to how Windows handles physical device writes.

mariobalanica, 2018-07-08:
Setting the FAT32 boot MBR partition as active also fixes the first boot error.
The only issue is that after Windows finishes the configuration it will hang at a black screen and restart if any USB device is plugged in. After that it will boot normally.

andreiw, 2018-07-08:
So there is no need for a hybrid GPT?

A

mariobalanica, 2018-07-09:
No.

closed 2019-10-20 by andreiw