andreiw/RaspberryPiPkg #40: Can support Aarch64 debian on Pi 3B with tf card.

aiminickwong, 2018-06-20:
I have test with, Aarch64 debian can start normal, but can not start with this.

aiminickwong, 2018-06-21:
@Googulator I don't know much about  SBSA, I used debian official repo with debootstrap tools and build the rootfs, built the kernel add uefi supoort .  All of this with CROSS_COMPILE.
Do not use RaspberryPiPkg Boot loader, everything is ok with normal boot and RPi office loader ,but cant not boot with RaspberryPiPkg.

thchi12, 2018-06-21:
Maybe your debian doen't have an efi bootloader?

aiminickwong, 2018-06-21:
@thchi12 what you mean about efi bootloader?  it's uefi support for kernel?  I'm quite sure I have make UEFI support  in Kernel build. So I don't known the efi bootloader what's it could you tell me, please :)  Thank you.

aiminickwong, 2018-06-21:
@thchi12 I have make EFI Boot Stub In kernel and let pi support start direct from kernel in EFI mode , cuz grub-uefi I don't know how to used to boot the kernel with the Tiano Core so I have just test EFI Boot Stub, but it's not work On Pi 3b+ （：

andreiw, 2018-06-21:
alright stay tuned...

when you boot the regular way, do you override the dtb? What is the config.txt when you boot Debian directly?

A

andreiw, 2018-06-22:
Okay so you had to build your own kernel.

Can you qualify what "does not work" mean? What do you see on the serial console? "Exiting boot stub" message from the kernel and that's it?

I noticed that on the Pi Debian image there's a DTB distributed.  Can you follow the directions in https://github.com/andreiw/RaspberryPiPkg#custom-device-tree to make sure that DTB is used to boot your Debian kernel?

andreiw, 2018-06-22:
Also follow https://github.com/andreiw/RaspberryPiPkg#custom-bootargs to make sure the correct command line is passed.

andreiw, 2018-06-22:
I can help you debug this further, but I don't have the time to install Debian and then build a custom kernel for the Pi with EFI stub support. You're welcome to attach your built kernel here as an attachment, though, if you would like me to take a look.

aiminickwong, 2018-06-22:
@andreiw I'll be back when I test again and collect some necessary information. Thank you.

aiminickwong, 2018-06-22:
I forget report,  before testing, I can not normally boot when power on I get the Rainbow screen, nothing go on with out any logs,  I rename the kernel8.img to k8 and boot to EFI shell and input "k8" to testing BOOT.
Now I have testing again, and with  UART tty console.

#### 1st is boot screen error, this info can not display on UART tty console screen.
----
[image]

#### UART tty console error when boot k8:
----
Shell> k8
FSOpen: Open '\k8' Success
FSOpen: Open '\k8' Success
FSOpen: Open '\k8' Success
FSOpen: Open '\k8' Success
[Security] 3rd party image[0] can be loaded after EndOfDxe: VenHw(58ABD787-F64D-4CA2-A034-B9AC2D5AD0CF)/HD(1,MBR,0x65FAAEF4,0x2000,0x1F800)/\k8.
InstallProtocolInterface: 5B1B31A1-9562-11D2-8E3F-00A0C969723B 35FF3240
Loading driver at 0x00032901000 EntryPoint=0x000333C695C
Loading driver at 0x00032901000 EntryPoint=0x000333C695C
InstallProtocolInterface: BC62157E-3E33-4FEC-9920-2D3B36D750DF 35FF3E18
ProtectUefiImageCommon - 0x35FF3240
  - 0x0000000032901000 - 0x0000000000F4F000
InstallProtocolInterface: 752F3136-4E16-4FDC-A22A-E5F46812F4CA 3B3FF5B8
EFI stub: Booting Linux Kernel...
EFI stub: EFI_RNG_PROTOCOL unavailable, no randomness supplied
ConvertPages: range 80000 - FCFFFF covers multiple entries
EFI stub: Using DTB from configuration table
EFI stub: Exiting boot services and installing virtual address map...
Exiting UEFI Boot Services!
----> Exiting UEFI and booting OS kernel! <----
SetUefiImageMemoryAttributes - 0x0000000037260000 - 0x0000000000040000 (0x0000000000000008)
SetUefiImageMemoryAttributes - 0x0000000033EF0000 - 0x0000000000040000 (0x0000000000000008)
SetUefiImageMemoryAttributes - 0x0000000033EA0000 - 0x0000000000040000 (0x0000000000000008)
SetUefiImageMemoryAttributes - 0x0000000033E30000 - 0x0000000000040000 (0x0000000000000008)
SetUefiImageMemoryAttributes - 0x0000000033D90000 - 0x0000000000040000 (0x0000000000000008)
SetUefiImageMemoryAttributes - 0x0000000033C10000 - 0x0000000000040000 (0x0000000000000008)
SetUefiImageMemoryAttributes - 0x0000000033B70000 - 0x0000000000040000 (0x0000000000000008)

#### tty start boot log:
----
[UART tty console error.txt](https://github.com/andreiw/RaspberryPiPkg/files/2128622/UART.tty.console.error.txt)

#### config:
----
arm_control=0x200
enable_uart=1
armstub=RPI_EFI.fd
disable_commandline_tags=1
device_tree_address=0x8000
device_tree_end=0x10000
device_tree=bcm2710-rpi-3-b-plus.dtb
----
#### cmdline:
--------
dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes net.ifnames=0 rootwait
----

#### my whole boot partition zip pkg include kernel file.
----
[boot.zip](https://github.com/andreiw/RaspberryPiPkg/files/2128641/boot.zip)

aiminickwong, 2018-06-22:
#### normal boot log with out EFI:
----
pi@raspbian:~$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.50-OPENFANS+CecOS-v8 (root@debian) (gcc version 6.3.0 20170516 (Debian 6.3.0-18)) #1 SMP PREEMPT Fri Jun 22 21:26:25 CST 2018
[    0.000000] Boot CPU: AArch64 Processor [410fd034]
[    0.000000] Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] cma: Reserved 8 MiB at 0x000000003ac00000
[    0.000000] On node 0 totalpages: 242688
[    0.000000]   DMA zone: 3792 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242688 pages, LIFO batch:31
[    0.000000] percpu: Embedded 22 pages/cpu @fffffff6fab7a000 s50968 r8192 d30952 u90112
[    0.000000] pcpu-alloc: s50968 r8192 d30952 u90112 alloc=22*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: enabling workaround for ARM erratum 845719
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 238896
[    0.000000] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=546dba46-02 rootfstype=ext4 elevator=deadline fsck.repair=yes net.ifnames=0 rootwait
[    0.000000] PID hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes)
[    0.000000] Memory: 928368K/970752K available (8316K kernel code, 978K rwdata, 2672K rodata, 2944K init, 720K bss, 34192K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     modules : 0xffffff8000000000 - 0xffffff8008000000   (   128 MB)
[    0.000000]     vmalloc : 0xffffff8008000000 - 0xffffffbebfff0000   (   250 GB)
[    0.000000]       .text : 0xffffff8541280000 - 0xffffff8541aa0000   (  8320 KB)
[    0.000000]     .rodata : 0xffffff8541aa0000 - 0xffffff8541d40000   (  2688 KB)
[    0.000000]       .init : 0xffffff8541d40000 - 0xffffff8542020000   (  2944 KB)
[    0.000000]       .data : 0xffffff8542020000 - 0xffffff8542114a00   (   979 KB)
[    0.000000]        .bss : 0xffffff8542114a00 - 0xffffff85421c8ce8   (   721 KB)
[    0.000000]     fixed   : 0xffffffbefe7fb000 - 0xffffffbefec00000   (  4116 KB)
[    0.000000]     PCI I/O : 0xffffffbefee00000 - 0xffffffbeffe00000   (    16 MB)
[    0.000000]     vmemmap : 0xffffffbf00000000 - 0xffffffc000000000   (     4 GB maximum)
[    0.000000]               0xffffffbfdb000000 - 0xffffffbfdbed0000   (    14 MB actual)
[    0.000000]     memory  : 0xfffffff6c0000000 - 0xfffffff6fb400000   (   948 MB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 28148 entries in 110 pages
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000]  Tasks RCU enabled.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000231] Console: colour dummy device 80x25
[    0.001073] console [tty1] enabled
[    0.001114] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=19200)
[    0.001160] pid_max: default: 32768 minimum: 301
[    0.001506] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes)
[    0.001553] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes)
[    0.002560] Disabling memory control group subsystem
[    0.007107] ASID allocator initialised with 32768 entries
[    0.009105] Hierarchical SRCU implementation.
[    0.011355] EFI services will not be available.
[    0.013172] smp: Bringing up secondary CPUs ...
[    0.020373] Detected VIPT I-cache on CPU1
[    0.020441] CPU1: Booted secondary processor [410fd034]
[    0.027479] Detected VIPT I-cache on CPU2
[    0.027526] CPU2: Booted secondary processor [410fd034]
[    0.034595] Detected VIPT I-cache on CPU3
[    0.034637] CPU3: Booted secondary processor [410fd034]
[    0.034775] smp: Brought up 1 node, 4 CPUs
[    0.034903] SMP: Total of 4 processors activated.
[    0.034933] CPU features: detected feature: 32-bit EL0 Support
[    0.034962] CPU features: detected feature: Kernel page table isolation (KPTI)
[    0.038345] CPU: All CPU(s) started at EL2
[    0.038395] alternatives: patching kernel code
[    0.039685] devtmpfs: initialized
[... 211 lines cut ...]
[    4.219688] systemd[1]: [/etc/systemd/system/rc-local.service:11] Support for option SysVStartPriority= has been removed and it is ignored
[    4.252165] hid-generic 0003:1C4F:0002.0002: input,hidraw1: USB HID v1.10 Device [SIGMACHIP USB Keyboard] on usb-3f980000.usb-1.3/input1
[    4.257967] systemd[1]: Listening on Journal Socket.
[    4.287205] systemd[1]: Started Dispatch Password Requests to Console Directory Watch.
[    4.307228] systemd[1]: Listening on udev Kernel Socket.
[    4.325267] systemd[1]: Set up automount Arbitrary Executable File Formats File System Automount Point.
[... cut ...]
[    5.994209] usbcore: registered new interface driver brcmfmac
[    6.217400] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04
[    6.238722] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2 Data: 9.10.105 Compiler: 1.29.4 ClmImport: 1.36.3 Creation: 2018-03-09 18:56:28
[    8.226377] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    8.233893] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    8.260419] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready

andreiw, 2018-06-22:
did you build your own kernel? Please share it

A

aiminickwong, 2018-06-22:
@andreiw I have upload https://github.com/andreiw/RaspberryPiPkg/files/2128641/boot.zip

aiminickwong, 2018-06-22:
@andreiw   zip pkg above include whole boot part also with kernel.img :)

aiminickwong, 2018-06-22:
@andreiw  and I used the latest RaspberryPiPkg bin file.

aiminickwong, 2018-06-23:
@andreiw 
Hi @andreiw  I'm go on testing, and now I have a very interesting test result.
3 different result, now I show the config.txt first:
----

arm_control=0x200
enable_uart=1
armstub=RPI_EFI.fd
disable_commandline_tags=2
device_tree_address=0x8000
device_tree_end=0x10000
device_tree=bcm2710-rpi-3-b-plus.dtb

----

#### Testing1:
----
Keep kernel file named kernel.img or kernel8.img
booting...
**Result:** Rainbow screen, stoped and no logs output.

#### Testing2:
----
change kernel file name to k8
booting to shell >
input k8 then press enter.
**Result:**  Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)

#### Testing3:
----
change kernel file name to k8
booting..
Press "ESC" to menu
chose "Boot Maintenace Manager" -> "Boot From File"  -> Chose "boot,[VenHw(...) /HD(1,MBR,...)]" -> chose file "k8" -> Press "Enter"
**Result:**
Success. System can boot normally and can find the efi and dmi.
PS: At Testing3 in config.txt, should set "disable_commandline_tags=2"  , if set "disable_commandline_tags=1" can not start.

----
The way to use "Boot From File" can start normally, other ways was broken.
This is really very, very, very strange. Any suggestions? I don't know too much about "Boot From File" mode. Why is this?

andreiw, 2018-06-24:
- Testing1 is not supposed to work (it's documented in the README.md, too!) disable_commandline_tags is also documented.

To understand Testing2 and Testing3 you need to understand how the kernel EFI stub works. The kernel EFI stub unconditionally overrides the device tree command line *IFF* a command line was provided by EFI (through EFI_LOADED_IMAGE_PROTOCOL->LoadOptions, if LoadOptionsSize != 0).

- Testing2 will not work unless you provide the *whole* command line after manually invoking k8.  This is because the UEFI shell will set the command line to include the program name, so "just" invoking k8 will in effect set the command line to be "k8". You were expecting an empty command line, but that's not how it works.

- Testing3 works because the LoadOptions are NULL and LoadOptionsSize is 0 (unless you manually set it to be something else). The EFI stub then doesn't override the FDT command line, and you end up using whatever you put into cmdline.exe.

My suggestion to you - name the kernel as \efi\boot\bootaa64.efi or better yet, use a bootloader like GRUB.

closed 2018-06-24 by andreiw

aiminickwong, 2018-06-25:
@andreiw That's great, it's very easy now. when just mv  /boot/kernel8.img  or  /boot/k8 to  /boot/efi/boot/bootaa64.efi ,  everything works now,  thanks  andreiw!