andreiw/RaspberryPiPkg #41: Booting this UEFI on the Raspberry Pi 3 B+

stefan2225, 2018-06-23:
The UEFI is fully compatible with the RPi 3B+ and I managed to install Fedora 28 Minimal with a custom DTB on an external USB Stick. I just needed to download the latest bootcode.bin, start.elf, fixup.dat. Additionally I needed to add the files fixup_cd.dat, fixup_db.dat, fixup_x.dat and start_cd.elf, start_db.elf, start_x.elf to get the RPi booting.

andreiw, 2018-06-23:
The current prebuilts already have the 3B+ compatible bootcode/start/fixup. Did that not work for you?

Thanks for the Fedora report.

A

stefan2225, 2018-06-23:
I just get the green LED blinking 4-5 on the RPi if I just use the already provided bootcode/start/fixup without the cd/db/x files. I also got no HDMI output, not even an rainbow screen. BTW I'm using the latest release build.

andreiw, 2018-06-23:
interesting, ok I’ll refresh those bits...

Can you provide the md5 sums for the known good VC firmware files (boot code, fixup and the rest) 

A

stefan2225, 2018-06-24:
Is it normal that Fedora detects when I remove the SD Card and when i reinsert it? I thought I have seen in a issue that that doesn't work.

thchi12, 2018-06-24:
Maybe that issue is talking about detecting card removal in UEFI?

satmandu, 2018-06-27:
@stefan2225 Which dtb did you use for the rpi 3b+?

stefan2225, 2018-06-28:
@satmandu I extracted the DTB from a Fedora 28 image with U-Boot.

Googulator, 2018-07-05:
There is a config.txt option that switches between the normal, cd, db or x versions. If you set that to the x version, it won't even look for the normal one.

andreiw, 2018-07-06:
Yeah don't do that.