andreiw/RaspberryPiPkg #42: Power Problem.

aiminickwong, 2018-06-25:
I found a problem, when shutdown the system or  command  "init 0" ,  system will  stoped .
It's looks like system level has shutdown ,but soon a later system will be auto start, I don't know why , may be it's a issue.

aiminickwong, 2018-06-25:
3b it's ok , just in 3b+

aiminickwong, 2018-06-25:
@andreiw how to change it to really power off?

andreiw, 2018-06-25:
right now there’s no way

Low prio work but I’ll get it done eventually

aiminickwong, 2018-06-25:
@andreiw  expect！ When you done this , this firmware will be perfect!

satmandu, 2018-07-06:
I think there is an interesting difference here between what the standard booted images do compared to what an image booted through Tiano Core does. With the standard booted images, doing a powerdown will not power cycle or soft-reset. With Tiano Core, halting does power cycle (or is it a soft-reset?)

(Am I wrong here?) This behavior is expected, so much so that there's provision for adding a soft-reset switch by jumping the P6 header as per here (apparently this was added in rev 2 of the original pi and has carried forward to newer versions.)

 https://raspi.tv/2012/making-a-reset-switch-for-your-rev-2-raspberry-pi

It's nice to be able to tell the system to shutdown, so you can remove power or swap an SD card without risking corruption, especially for headless systems. Otherwise you have to time the shutdown command with pulling power as soon as the system has started to power cycle come back up.

Tl;DR: Expected behavior on sending shutdown is to shift the pi into a low power mode which can only be recovered from via power cycle or shorting header P6. With Tiano Core sending shutdown appears to end in a warm-reset.

andreiw, 2018-07-06:
Quick workaround: use “halt” instead of “poweroff”.

I don’t disagree with you, I’ll have a fix eventually, but I do think it’s Linux that should cope with PSCI POWER_OFF failing by halting instead. Sigh.

A

satmandu, 2018-07-06:
Here's the officially suggested solution https://github.com/raspberrypi/firmware/issues/1016#issuecomment-403139219 , which is to "write 0x555 (plus the usual password) to the RSTS register,"  _before_ using the watchdog timer to reset the device. 

That would be a minor modification to the rpi3_system_reset function in https://github.com/andreiw/raspberry-pi3-atf/blob/master/plat/rpi3/rpi3_pm.c, making rpi3_system_off a special case inside rpi3_system_reset, no?

Or was this what was happening in the originally without a shutdown function before you implemented the https://github.com/andreiw/raspberry-pi3-atf/commit/b1f5b0343b10562685dbe603585cba565a4bb54a commit?

andreiw, 2018-07-06:
> Here's the officially suggested solution raspberrypi/firmware#1016 (comment) , which is to "write 0x555 (plus the usual password) to the RSTS register," before using the watchdog timer to reset the device.
> Or was this what was happening in the originally without a shutdown function before you implemented the andreiw/raspberry-pi3-atf@b1f5b03 commit?

Sure, I can do that

A