andreiw/RaspberryPiPkg #44: Im newbie pls help

moltenteam, 2018-06-30:
Hello, im trying to install win 10 on rpi. I have bootable USB with win 10 and i cannot boot from this USB. If I load into the setup and select boot manager then i clicked on the USB port which i plugged on win 10 usb.

stefan2225, 2018-06-30:
You can't just boot an normal Windows 10 USB made with the media creation tool. It's for x86_64 and not arm64.

moltenteam, 2018-06-30:
Can anybody tell me how much storage will consume the ADK on the windows host?

moltenteam, 2018-06-30:
Can anybody provide disc image file with windows 10 arm64 installation?

stefan2225, 2018-06-30:
You need to make one yourself with the drivers added for the RPi 3. There are some tutorials/guides on how to do this on YouTube and various other platforms.

nikhildatta, 2018-06-30:
This is the tutorial:
https://onedrive.live.com/redir?resid=C8B0DBB75540E8D5!1187054&authkey=!AIWR8mCQAfVuGOo&ithint=file%2cdocx
Check this link for downloading win10: https://uup.rg-adguard.net/

moltenteam, 2018-07-02:
thank you nikhildatta

closed 2018-07-02 by moltenteam