andreiw/RaspberryPiPkg #45: 2018May18 GCC4.9 WoA UEFI Build

SparrOSDeveloperTeam, 2018-06-30:
I have tried to boot WoA using the build above and for some reason I keep getting a
```
--> Exiting UEFI Booting OS Kernel <--
```
Message along with a bar below.

I have tried booting the following files:

```
EFI\boot\bootaa64.efi
EFI\Microsoft\Recovery\BCD
EFI\Microsoft\Boot\bootmgfw.efi
EFI\Microsoft\Boot\bootmgr.efi
EFI\Microsoft\Boot\BCD
EFI\Microsoft\Boot\memtest.efi
```

and same error with all except memtest, and BCD.

I used BCDBOOT to create a uefi bios and the efi files. I think I may try using the efi files from the disc and different WoA builds. I have integrated your drivers into the wim archive and i edited the bcd so it doesnt use integrity checks and does a faster boot.

SparrOSDeveloperTeam, 2018-07-01:
Hi I am still having problems with booting Windows 10. I've been at it all this weekend. I have been able to write the install.wim to sd card, force installation of unsigned drivers, used bcdboot to make UEFI files, and I am able to get to Windows Splash. Unfortunately I keep getting a `DRIVER_POWER_STATE_FAILURE` after Windows trys booting.

[image]

I am using Windows 10 Version 1703 Arm64 Home Single Language and my Raspberry Pi 3 B model. Please help.

thchi12, 2018-07-01:
Don't plug any usb devices while booting.
BTW.....you have 1703 arm64??

SparrOSDeveloperTeam, 2018-07-01:
Yea it required a few hours of research on the internet to get build 1703 Arm64. Also I think that solved my prob because that started the setup info.

[image]

I am not gonna close this issue yet because I feel something may happen.

SparrOSDeveloperTeam, 2018-07-02:
Oh man I forgot about the registry hack. The file that skips setup. Well other than that I think that will be all... for now.

SparrOSDeveloperTeam, 2018-07-02:
OMG Thank you

[image]

I have waited to see this screen in a while. What I had to do was just get Windows 10 booting w/o any PnP devices connected, `regedit regfile.reg`, restart and now I am here. Thank you. Issue resolved.

closed 2018-07-02 by SparrOSDeveloperTeam

thchi12, 2018-07-02:
Could you provide the link to 1703 version? I want a try plz

thchi12, 2018-07-02:
Ok..I want only the way ti get to that..
There're 15063.xx links on MDL's uup dump.
But all links lack edition_lang.esd and cannot be converted.

SparrOSDeveloperTeam, 2018-11-05:
Its been a while since I've used Windows 10 on my RPi. 

> Please don’t share build links here, thanks

 I'm not.

Anyway it's been months since I've used Win10 on my Pi. It seems like a cool concept but I'm hoping we can find a way to emulate ReactOS on RPi. I feel like if we can get ReactOS running on a Raspberry Pi then it would be more stable than Windows 10.

BenNottelling, 2018-11-05:
Really? You think ReactOS would be more stable? I don't think so ...
Is there even an arm64 build of ReactOS?

SparrOSDeveloperTeam, 2018-11-05:
No its just ARM. I may be dreaming a little too big but I think it could be possible to port ReactOS to RPi, however idk how well it would perform. If I can't get ReactOS to work on RPi then I may end up making a Raspberry Pi image instead and get it to look and work similar to ROS.

SparrOSDeveloperTeam, 2018-11-05:
It could work. The only problem is ReactOS has less drivers for it though so it could take longer to get it working on the Raspberry Pi. I think the Broadcom driver should work. Not sure about the RPi SDHC though. I'm mainly concerned about whether it will boot or not. I'm not currently working on the project because I'm working on a custom Windows 10 edition that installs ClassicShell and 8GadgetPack to make it feel more like Windows 7. When I am done tinkering around with Windows 10 then I'll start getting to work on ROS.

SparrOSDeveloperTeam, 2018-12-27:
@thchi12 I now have a Repo dedicated to ReactOS on ARM [here](https://github.com/sparrdrem/rpi-roa)