andreiw/RaspberryPiPkg #47: pseudo-NVRAM not working (only works on pre-built images)

esnowberg, 2018-07-05:
If I use one of the pre-built images the pseudo-NVRAM works fine.  However, if I follow the build instructions within the readme.   The new RPI_EFI.fd image I built works, with the exception of the pseudo-NVRAM.  Are there additional build steps to get that to work?

andreiw, 2018-07-05:
What are the exact steps and what behavior are you seeing exactly? An attached serial log from a debug build will be useful.

A

esnowberg, 2018-07-06:
I'm trying to set the System BootOrder and add a Boot Entry.  It will persist with your prebuilt image.  But it doesn't with the one I built.

Debug log attached.
[nvram-problem.txt](https://github.com/andreiw/RaspberryPiPkg/files/2168835/nvram-problem.txt)

andreiw, 2018-07-06:
Fun.

- Are there any other media present that may have an RPI_EFI.FD file on them?
- Can you build using GCC 4.9 or GCC5 (I know, I know..)
- After creating the boot entry, what happens when you select "reset the system" (can you give me a full log with that?)

esnowberg, 2018-07-06:
There isn't any other media present.  The only thing connected is the micro sd card and the serial cable.

The BOOTAA64.EFI program is really fbaa64.efi, so it creates the boot entries and then jumps to shim. This time once shim loaded GRUB2  Instead of booting into Linux, I rebooted from the GRUB2 shell.  This brought me back to UEFI.  I was then going to reset the system.  But first I looked at the Boot Manager Menu and saw my variable set.  Instead of resetting the system, I just booted into Linux.  Now the two variables persist.

To reproduce the problem if I do:
UEFI -> BOOTAA64.EFI (which sets variables) -> shim -> GRUB2 -> Linux
The variables created do not persist.

If I do:
UEFI -> BOOTAA64.EFI (which sets variables) -> shim -> GRUB2 -> reboot 
The variables persist.

andreiw, 2018-07-06:
Yeah I see now.

Okay, thanks for the report. I’ll share a test build with you soon (attached to this ticket)

A

andreiw, 2018-08-24:
Soon. Sorry, a little overwhelmed right now with other stuff.

closed 2018-09-18 by andreiw