andreiw/RaspberryPiPkg #48: compiling on native aarch64 host

stefan2225, 2018-07-06:
Hello,

I tried to compile this on an aarch64 host but I get some errors I can't seem to resolve. Seems like many other people have this issue but with compiling edk2 for an x86_64 host. Would be nice if you can tell me what I did wrong (I attached the log to this issue).

Thanks,
Stefan.
[log.txt](https://github.com/andreiw/RaspberryPiPkg/files/2170803/log.txt)

andreiw, 2018-07-06:
This has nothing to do with an AArch64 host.

You’re missing uuid-dev or whatever package providing libuuid lib/headers on your system.

'/home/******/git/edk2/BaseTools/Source/C/GenFv' gcc -o ../bin/GenFv GenFv.o GenFvInternalLib.o -L../libs -lCommon -luuid /usr/bin/ld: cannot find -luuid collect2: error: ld returned 1 exit status make[2]: *** [../Makefiles/app.makefile:24:
A

stefan2225, 2018-07-06:
Weird, I have installed uuid-dev both on my normal pc and on the aarch64 host but It only compiled on my normal pc. sorry

andreiw, 2018-07-06:
Btw edk2 build only works on little endian hosts. Just in case you decide to try aarch64be...

A

stefan2225, 2018-07-06:
Okay, I just ran lscpu on it and it tells me that it's little endian..

stefan2225, 2018-07-06:
Maybe it is something with the kernel, since it's mainline..?

andreiw, 2018-07-06:
I'm confused about your question.

Your compilation issues stem from the build system not finding libuuid.

A

stefan2225, 2018-07-06:
Yes, but I mean that I have them already installed

andreiw, 2018-07-06:
Can you build this?

https://gist.github.com/yoggy/4483031

stefan2225, 2018-07-06:
No, I get a error message instead:

/tmp/ccHcUT1C.o: In function `main':
uuid_test.c:(.text+0x14): undefined reference to `uuid_generate_time_safe'
uuid_test.c:(.text+0x20): undefined reference to `uuid_unparse_lower'
uuid_test.c:(.text+0x40): undefined reference to `uuid_parse'
uuid_test.c:(.text+0x4c): undefined reference to `uuid_compare'
uuid_test.c:(.text+0x74): undefined reference to `uuid_parse'
uuid_test.c:(.text+0x80): undefined reference to `uuid_compare'
uuid_test.c:(.text+0x9c): undefined reference to `uuid_is_null'
uuid_test.c:(.text+0xb8): undefined reference to `uuid_clear'
uuid_test.c:(.text+0xc0): undefined reference to `uuid_is_null'
collect2: error: ld returned 1 exit status

I'll try reinstalling the libuuid packages.

andreiw, 2018-07-08:
Not a RaspberryPiPkg issue

closed 2018-07-08 by andreiw