andreiw/RaspberryPiPkg #49: Windows wont run

ham1255, 2018-07-16:
So get error that need reg file so i done it then windows did crAsh then reboot then getting bad state error even there no usb plugged in

ham1255, 2018-07-16:
Then i try did all steps again then new error show up 
Windows setup could bot configer windows to run on this computer

thchi12, 2018-07-17:
You don't need the reg hack
Making your FAT partition active solves the setup error.

ham1255, 2018-07-17:
Making alll partitions as fat?

thchi12, 2018-07-18:
No..
You've made two partitions. Just make the partition which has boot files active. （ It should be a FAT16 or FAT32 partition. ）

thchi12, 2018-07-18:
Then make the fat32 partition active.

ham1255, 2018-07-19:
Is there sound drivers yet?

driver1998, 2018-07-20:
We do, checkout github.com/driver1998/bsp if you haven’t know yet.

发送自我的 Windows 10 手机

ham1255, 2018-07-20:
@driver1998 is there ethernet driver too?

andreiw, 2018-07-22:
Not a uefi issue. Use issue 12 instead

andreiw, 2018-07-22:
https://github.com/andreiw/RaspberryPiPkg/issues/12

closed 2018-07-22 by andreiw