andreiw/RaspberryPiPkg #53: SMBIOS tables should only be using RTS data

andreiw, 2018-07-27:
Validate that no BS data is referenced by SMBIOS tables.

@driver1998 noticed that a few of the strings were empty when seen from Windows. To be investigated.

andreiw, 2018-09-14:
@driver1998 do you remember which strings were not working? is it once in a while or always?

driver1998, 2018-09-14:
If you check that registery key I mentioned at that time, you'll see that the whole key is empty. There are offical MS docs about that key.

driver1998, 2018-09-14:
That is HKLM\Hardware\Description\BIOS

driver1998, 2018-09-14:
MS docs about this
http://download.microsoft.com/download/5/D/6/5D6EAF2B-7DDF-476B-93DC-7CF0072878E6/SMBIOS.doc
It is old, but I think it doesn't really changed, as both the MS UEFI for PI and the UEFI of my Lenovo laptop all have correct values there.

driver1998, 2018-09-14:
Well the key is apparently HKLM\Hardware\Description\System\BIOS now, I don't have a Windows machine nearby to check though.
It is used by Windows and some application to get the SKU and model number of the machine.

andreiw, 2018-09-14:
Yes, I found the key and it is empty.

Is this a regression? i.e. has it ever worked in this firmware?

andreiw, 2018-09-14:
[image]

I have a fix. It was a regression :(. Will update binaries today.

closed 2018-09-14 by andreiw

andreiw, 2018-09-14:
Fixed in https://github.com/andreiw/RaspberryPiPkg/tree/master/Binary/prebuilt/2018Sep14-GCC5