andreiw/RaspberryPiPkg #54: How do I boot into windows?

jordanwilliams1, 2018-07-30:
I Followed this tutorial:
https://www.youtube.com/watch?v=c0VqVm8X_zQ

But straight after this screen:
https://i.gyazo.com/dad8ca6e43391c313a3daffb1933faf6.png

I Get this:
https://i.gyazo.com/89d0d40ba0c23a0b76713afb4f7a2c6a.jpg

andreiw, 2018-07-30:
Can you type the following into the efi Shell?

fs0: <press enter>
dir efi\boot

A

cdps82, 2018-08-03:
And when I go to 

cd efi\boot and type 
bootx64.efi 

I get: Command Error Status: Unsupported.

Can anyone help?

andreiw, 2018-08-03:
You’ve got the wrong (x86) Windows bits. the right bits have a file bootaa64.efi

A

cdps82, 2018-08-03:
Could the reason be the language choice? As I am sure I am choosing the right version (17134.1_cs-cz_arm64_Professional). Thanks

cdps82, 2018-08-03:
OK, JFYI: confirming that 3 different language mutations are the cause. EN-US is fine.

closed 2018-08-13 by andreiw