andreiw/RaspberryPiPkg #61: Route PL011 UART to GPIOs 32 and 33 for Bluetooth

Googulator, 2018-09-11:
Currently GPIOs 32 and 33 are left in their default configuration, which is likely GPIO input. These pins are used on the Pi 3 to interface with the Bluetooth controller. To route the correct UART signals to them, they need to be configured as ALT3.

Additionally, GPIOs 30 and 31 appear to be used for Bluetooth PCM audio, requiring an ALT2 configuration.

andreiw, 2018-09-13:
Good idea, thanks, will expose this through chipset config and make it the default configuration.

andreiw, 2018-09-18:
@Googulator can you confirm if these routings are the same across Pi3 and Pi3B+?

Googulator, 2018-09-24:
Yes, it's the same on all Pis equipped with Bluetooth. Probably even on the Zero W.

Googulator, 2018-09-24:
BTW, why are you always rebasing the README on top? Makes it hard to see at a glance when the repo changes.

closed 2020-02-19 by andreiw