andreiw/RaspberryPiPkg #64: Win10 shutdown does not soft-off

andreiw, 2018-09-14:
The screen goes blank, but not black, and the green led doesn't blink, meaning that Windows never called PSCI to power-off.

andreiw, 2018-09-14:
poweroff -f does the right thing in Linux.... possible some ACPI-related shenanigans

driver1998, 2018-09-14:
Speaking of by blinking LEDs, when I install the ported RPIQ driver, the green LED keeps blinking all the time. But the driver does work though (I can send ioctls to it for mailbox stuff).

andreiw, 2018-09-14:
Well, it's either the driver or the GPU...

andreiw, 2018-09-14:
I really wish ACPI added an OpRegion for SMCs. Grrr.

I think I could get away with a \_PTS method to basically implement https://github.com/andreiw/raspberry-pi3-atf/blob/11c9a950ea89a753e473c8b66d6cfe23e9ce7e42/plat/rpi3/rpi3_pm.c#L220

mariobalanica, 2019-02-15:
Now it turns off properly. 
The issue was the faulty Windows USB driver.

andreiw, 2019-02-15:
Ok, good to know

closed 2019-02-15 by andreiw