andreiw/RaspberryPiPkg #67: error ?

robinjoo1, 2018-09-15:
i dont know if this is the right place to place this error 
seems it has an error applying driver files
2018-09-15 21:18:57.4045|TRACE|WoR.MainWindow| WoR started!
2018-09-15 21:18:57.4784|DEBUG|WoR.MainWindow| Loaded language file: English
2018-09-15 21:19:02.9016|DEBUG|WoR.MainWindow| Loading drives...
2018-09-15 21:19:02.9357|DEBUG|WoR.MainWindow| Detected drives: 
Disk 0 - Samsung SSD 860 EVO 500GB - Fixed hard disk media - 465GB
2018-09-15 21:20:57.3568|DEBUG|WoR.MainWindow| Loading drives...
2018-09-15 21:20:57.4390|DEBUG|WoR.MainWindow| Detected drives: 
Disk 0 - Samsung SSD 860 EVO 500GB - Fixed hard disk media - 465GB
Disk 1 - SDXC Card - Removable Media - 119GB
2018-09-15 21:21:00.4007|DEBUG|WoR.MainWindow| Selected drive: 
Disk 1 - SDXC Card - 119GB
2018-09-15 21:45:15.9294|DEBUG|WoR.MainWindow| Selected image (ISO): 17672.1000.180511-1445.RS_PRERELEASE_CLIENTCORE_OEMRET_A64FRE_EN-US.ISO
2018-09-15 21:45:16.0248|DEBUG|WoR.Image| Mounting ISO image...
2018-09-15 21:45:23.0984|DEBUG|WoR.Image| ISO file mounted on: I:\
2018-09-15 21:45:23.1231|DEBUG|WoR.ProcessCmd| Launch "dism.exe" with the following arguments: /Export-Image /SourceImageFile:I:\sources\install.wim /SourceIndex:1 /DestinationImageFile:"C:\Users\robin de bruin\Desktop\raspberry pi\WoR_Release0.2.2 - 1\temp\wim\install.wim"
2018-09-15 21:45:52.9186|DEBUG|WoR.ProcessCmd| dism.exe process output: 
Deployment Image Servicing and Management tool
Version: 10.0.17134.1

Exporting image

[                           1.0%                           ] 

[=                          2.0%                           ] 

[=                          3.0%                           ] 

[==                         4.0%                           ] 

[==                         5.0%                           ] 

[===                        6.0%                           ] 

[====                       7.0%                           ] 

[====                       8.0%                           ] 

[=====                      9.0%                           ] 

[=====                      10.0%                          ] 

[======                     11.0%                          ] 

[======                     12.0%                          ] 

[=======                    13.0%                          ] 

[========                   14.0%                          ] 

[========                   15.0%                          ] 

[=========                  16.0%                          ] 

[=========                  17.0%                          ] 

[==========                 18.0%                          ] 

[===========                19.0%                          ] 

[===========                20.0%                          ] 

[============               21.0%                          ] 

[============               22.0%                          ] 

[=============              23.0%                          ] 

[=============              24.0%                          ] 
[... 2389 lines cut ...]
Applying metadata to files: 120000 of 121521 (98%) done
Applying metadata to files: 120500 of 121521 (99%) done
Applying metadata to files: 121000 of 121521 (99%) done
Applying metadata to files: 121500 of 121521 (99%) done
Applying metadata to files: 121521 of 121521 (100%) done
Done applying WIM image.

2018-09-15 22:21:02.3536|DEBUG|WoR.ProcessCmd| wimlib-imagex.exe executed the task successfully!
2018-09-15 22:21:02.3935|DEBUG|WoR.MainWindow| The image has been applied to: S
2018-09-15 22:21:02.4165|DEBUG|WoR.MainWindow| Installing drivers...
2018-09-15 22:21:02.4294|DEBUG|WoR.ProcessCmd| Launch "dism.exe" with the following arguments: /image:S: /add-driver /driver:"C:\Users\robin de bruin\Desktop\raspberry pi\WoR_Release0.2.2 - 1\temp\drivers\" /recurse /forceunsigned
2018-09-15 22:21:18.4723|DEBUG|WoR.ProcessCmd| dism.exe process output: 
Deployment Image Servicing and Management tool
Version: 10.0.17134.1

Image Version: 10.0.17672.1000

Error: 2

Cannot find the specified file 'C:\Users\robin de bruin\Desktop\raspberry pi\WoR_Release0.2.2 - 1\temp\drivers" /recurse /forceunsigned'

The DISM log file can be found at C:\Windows\Logs\DISM\dism.log

2018-09-15 22:21:18.4723|ERROR|WoR.Exceptions.ProcessException| The dism.exe process has returned an error!
2018-09-15 22:21:18.5112|INFO|WoR.MainWindow| The installation could not be finished!

andreiw, 2018-09-15:
@mariobalanica

andreiw, 2018-09-15:
Mario is the WoR guy

robinjoo1, 2018-09-15:
> Mario is the WoR guy

ok since you mentioned his name will he see this ?
or should i also email him ?

andreiw, 2018-09-15:
He should see it

Отправлено с Apple Watch

mariobalanica, 2018-09-16:
https://www.worproject.ml/bugtracker/view.php?id=16

andreiw, 2018-09-18:
Great. I also added the worproject site/bug tracker to the readme.

closed 2018-09-18 by andreiw