andreiw/RaspberryPiPkg #70: Question: is the Raspberry Pi 2 rev. 1.2 (BCM2837) supported/working?

mk2soldier, 2018-09-20:
Great project! I was wondering if this UEFI port is compatible with the rev 1.2 of the Raspberry Pi 2 (which is based upon the BCM28367 SoC albeit with a lower default clock speed: 900 MHz)

Thank you!

andreiw, 2018-09-20:
You mean the 64-bit Pi 3 variant? It could, yes, but is not today - the firmware (both atf and uefi) need to be able to use either pl011 or the microuart serial ports, and today only the microuart is supported.

A

mk2soldier, 2018-09-20:
AFAIK apart from the CPU speed and the absence of wireless connectivity the Pi2 V1.2 and Pi3 should be virtually identical right?

andreiw, 2018-09-20:
Sorry, I mistyped, I meant the updated Pi 2 so we are talking about the same thing. The updated Pi 2 is basically a Pi 3 without BT/WLAN.

It should work, but I don’t have one to test (yet) and I don’t know if it requires use of PL011 UART or if it can continue to work with the miniUART.

A

mk2soldier, 2018-10-07:
Thank you for your reply, if you want I can do some testing: I have a rev. 1.2 RPi2 that I can use

andreiw, 2018-10-08:
Sure, thanks. Just try the latest release, just so I validate by expectations.

I expect that a) nothing will appear on screen b) serial output will not work.

(a) would be induced by a crash caused by (b), so the first thing to work on would be to see messages from ATF.

A

andreiw, 2018-10-09:
Ok, I ordered one.

mk2soldier, 2018-10-10:
I've downloaded the latest binary release, I've formatted the uSD (SanDisk 8GB) with MBR partition table and FAT32 filesystem, I've copied the content of `2018Oct1-GCC5/RELEASE/` (and also `2018Oct1-GCC5/DEBUG/` after the first try) to the root of the partition of the uSD card. When I boot the RPi all I can see is the colored rainbow screen, it remains there indefinitely.

Am I missing something there or is this the expected outcome?

IIRC the RPi 1.2 has the PL011 UART breaked out to pins 14/15 and the miniUART is confined in the SoC (not mapped to any external pins outside of the BCM2837 BGA package) or is it the other way around?

I've yet to try the serial console, I'll report back my findings

andreiw, 2018-10-10:
That’s the expected outcome. Most likely it crashes on the first microUART accesses, because the AUX block is not enabled by VC firmware at the time of the handover.

I bought one. Should arrive soon.

A

stefan2225, 2018-10-12:
@andreiw You can also set bluetooth to use the miniUART, instead of PL011 UART with "dtoverlay=pi3-miniuart-bt" in the config.txt, after that PL011 will be restored to pins 14/15 like on the pi 2 v1.2

andreiw, 2018-10-15:
I have a Pi 2 1.2 now. No ETA :)

mk2soldier, 2018-10-15:
I'm definitely looking forward to it :)

HoshiYamazaki, 2019-03-09:
Hi, is there something I can test? I own rPi 2 so I could spend free time on it :)

HoshiYamazaki, 2019-06-13:
He-llo?

andreiw, 2020-02-19:
If there will be, it will be with the upstream RPi3 code at https://github.com/tianocore/edk2-platforms/tree/master/Platform/RaspberryPi/RPi3

closed 2020-02-19 by andreiw