andreiw/RaspberryPiPkg #71: "Windows 10 ARM for Raspberry Pi" Drivers problem

Andrew752, 2018-09-20:
I heard that there is a problem with the sound card (there is no sound) and there is no connectivity (no WIFI or Bluetooth) can work on the raspberry pi so how we can fix this problem. when i heard that's i can use windows 10 on something very small like the Raspberry Pi, I was very interesting so i was thinking to buy the "Raspberry Pi 3 model b" next week and now i heard this bad news about the sound driver and the sd card driver and the bluetooth and wifi drivers. So how can we fix this problem. And thanks for reading my problem.[image]

andreiw, 2018-09-21:
The issues you highlight are not UEFI firmware issues and this project is UEFI firmware for the Pi, not Windows on the Pi. Sorry.

Specifically, RaspberryPiPkg is a UEFI firmware that can also, fairly incidentally, boot Windows on the Pi as well as Windows can boot today. You can boot other OSes, like Linux or FreeBSD/NetBSD, with this firmware or without, and they will work better than Windows does today.

Well, you have a few other choices. Choice 1 is to work on the driver support as part of the community of folks who are interested in that (eg USB and the accelerated display driver that seems to have bitrotted from the Pi 2 era)

Choice 2 is to complain to someone who cares at MSFT, and hope for official support.

Sorry dude, but Windows on the Pi is a community effort. There is no official support for Pi on regular 64-bit Windows 10, and I am not affiliated with Microsoft or the Pi Foundation. It got this far, and it’s pretty amazing that it did, but you get what you get. No guarantee of function or fitness and it might eat your cat.

A

Andrew752, 2018-09-21:
I'm really Sorry for that. I'm new in this field :) I don't knew the raspberry pi well, I'm still reading about it to knew how i can take my AI assistant (Works on windows only) any where. but thanks for your help and your support :).

closed 2018-09-21 by andreiw