andreiw/RaspberryPiPkg #74: found a typo on SMBIOS Vendor Information and more

stefan2225, 2018-09-22:
"https://github.org/andreiw/RaspberryPiPkg" should be "https://github.com/andreiw/RaspberryPiPkg" as the .org domain doesn't work.

It also says it's an Sony Laptop???? (below is the whole dmidecode output)

#########

SMBIOS 3.1.1 present.
Table at 0x37370000.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: https://github.org/andreiw/RaspberryPiPkg
        Version: Raspberry Pi 64-bit UEFI (0c77450 on 09/18/2018)
        Release Date: 09/18/2018
        ROM Size: 2048 kB
        Characteristics:
                Selectable boot is supported
                ACPI is supported
                Function key-initiated network boot is supported
                Targeted content distribution is supported
                UEFI is supported

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: Sony
        Product Name: 64-bit Raspberry Pi 3B+ (rev. A020D3)
        Version: 64-bit Raspberry Pi 3B+ (rev. A020D3)
        Serial Number: 0000000000000000
        UUID: 33695052-0000-0000-0000-000000000000
        Wake-up Type: Power Switch
        SKU Number: 0000000000A020D3
        Family: edk2

Handle 0x0002, DMI type 2, 17 bytes
Base Board Information
        Manufacturer: Sony
        Product Name: 64-bit Raspberry Pi 3B+ (rev. A020D3)
        Version: 64-bit Raspberry Pi 3B+ (rev. A020D3)
        Serial Number: 0000000000000000
        Asset Tag: None
        Features:
                Board is a hosting board
        Location In Chassis: 0000000000A020D3
        Chassis Handle: 0x0000
        Type: Motherboard
        Contained Object Handles: 0

Handle 0x0003, DMI type 3, 24 bytes
Chassis Information
        Manufacturer: Sony
        Type: Laptop
        Lock: Not Present
        Version: 64-bit Raspberry Pi 3B+ (rev. A020D3)
        Serial Number: 0000000000000000
        Asset Tag: None
        Boot-up State: Safe
        Power Supply State: Safe
        Thermal State: Safe
        Security Status: None
        OEM Information: 0x00000000
        Height: Unspecified
        Number Of Power Cords: Unspecified
        Contained Elements: 0
        SKU Number: Not Specified

Handle 0x0004, DMI type 4, 48 bytes
Processor Information
        Socket Designation: Socket
        Type: Central Processor
        Family: ARM
        Manufacturer: ARM
        ID: 00 00 00 00 00 00 00 00
        Version: BCM2837 ARMv8
        Voltage: 5.0 V 3.3 V 2.9 V
        External Clock: Unknown
        Max Speed: 1400 MHz
        Current Speed: 600 MHz
        Status: Populated, Enabled
        Upgrade: Other
        L1 Cache Handle: 0x0000
        L2 Cache Handle: 0x0000
        L3 Cache Handle: 0x0000
        Serial Number: 1.0
        Asset Tag: 1.0
        Part Number: 1.0
        Core Count: 4
        Core Enabled: 4
        Thread Count: 4
        Characteristics:
                64-bit capable
                Multi-Core
                Execute Protection
                Enhanced Virtualization

Handle 0x0005, DMI type 7, 27 bytes
Cache Information
        Socket Designation: Cache1
        Configuration: Enabled, Socketed, Level 3
        Operational Mode: Write Back
        Location: Internal
        Installed Size: 0 kB
        Maximum Size: 0 kB
        Supported SRAM Types:
                Burst
                Synchronous
        Installed SRAM Type: Burst Synchronous
        Speed: Unknown
        Error Correction Type: Multi-bit ECC
        System Type: Unknown
        Associativity: 2-way Set-associative

Handle 0x0006, DMI type 9, 17 bytes
System Slot Information
        Designation: SD Card
        Type: Other
        Current Usage: Available
        Length: Other
        Characteristics: Unknown
        Bus Address: 0000:00:00.0

Handle 0x0007, DMI type 16, 23 bytes
Physical Memory Array
        Location: System Board Or Motherboard
        Use: System Memory
        Error Correction Type: Unknown
        Maximum Capacity: 1 TB
        Error Information Handle: Not Provided
        Number Of Devices: 1

Handle 0x0008, DMI type 17, 40 bytes
Memory Device
        Array Handle: 0x0007
        Error Information Handle: Not Provided
        Total Width: Unknown
        Data Width: Unknown
        Size: 1024 MB
        Form Factor: Unknown
        Set: Unknown
        Locator: OS Virtual Memory
        Bank Locator: malloc
        Type: DRAM
        Type Detail: Unknown
        Speed: Unknown
        Manufacturer: OSV
        Serial Number: Not Specified
        Asset Tag: Not Specified
        Part Number: Not Specified
        Rank: Unknown
        Configured Clock Speed: Unknown
        Minimum Voltage: Unknown
        Maximum Voltage: Unknown
        Configured Voltage: Unknown

Handle 0x0009, DMI type 19, 31 bytes
Memory Array Mapped Address
        Starting Address: 0x00000000000
        Ending Address: 0x0003B3FFFFF
        Range Size: 948 MB
        Physical Array Handle: 0x0000
        Partition Width: 1

Handle 0x000A, DMI type 32, 11 bytes
System Boot Information
        Status: No errors detected

Handle 0xFEFF, DMI type 127, 4 bytes
End Of Table

mariobalanica, 2018-09-22:
http://www.sonypencoed.co.uk/case-studies/raspberry-pi/

stefan2225, 2018-09-22:
ok, thanks for the info, didn't now the rpi 3 was manufactured by sony

andreiw, 2018-09-24:
Thanks. I'll fix the .org->.com and report an "Embedded PC" chassis type.

There are/were other manufacturers (Embest, Egoman and Sony Japan). I've only seen Sony boards so far!

closed 2018-09-24 by andreiw