andreiw/RaspberryPiPkg #82: Provide (or document, if already exists) a way to disable passing FDT in EFI system table (required for Linux ACPI boot)

Googulator, 2018-10-01:
Linux ignores ACPI tables whenever an FDT is available, with no way to override it - even if that FDT is invalid. To test ACPI boot under Linux, we need a way to pass no FDT at all. Right now, this doesn't appear to be possible.

andreiw, 2018-10-01:
Yep, on top of my list. Thanks.

andreiw, 2018-10-04:
> Just checked the EBBR document - this is actually a compliance issue.
>
> 2.4.3 Configuration Tables
> A UEFI system that complies with this specification may provide the additional tables via the EFI Configuration Table.
> Compliant systems are required to provide one, but not both, of the following tables.
> • An Advanced Configuration and Power Interface [ACPI] table, or
> • a Devicetree [DTSPEC] system description
> As stated above, EBBR systems must not provide both ACPI and Devicetree tables at the same time.
> Systems that support both interfaces must provide a configuration mechanism to select either ACPI or Devicetree,
> and must ensure only the selected interface is provided to the OS loader.
>
> Long story short, EBBR requires ACPI/DT to be a strict either/or switch; providing both is an error, and operating systems may rightfully refuse to boot when both are provided.

Yep, well recognized. Stay tuned!

A

mrmajik45, 2018-10-31:
Stayin' tuned

driver1998, 2018-11-08:
The problem is, is no DT actually possible here? Considering that VC firmware rely on DT for hardware configuration.

andreiw, 2018-11-08:
@driver1998 yes, it will be possible - UEFI does not use the DT nor does it have to be exposed.

The problem is that Linux will not boot without it, unless someone explicitly adds support for the non-standard ACPI tables used by Windows (and by a certain other OS demoed yesterday ;-)).

So, is someone adding ACPI Linux support for Pi? Anyone have a vmlinux.efi for me to test with?

Sorry folks, I’ve been busy, but this is on top of my list.

A

driver1998, 2018-11-08:
Speaking of that other OS, I wonder if WoA can run on that, I tested WoA on KVM before, and it don't even load the kernel. (no kernel debug message, no display output, nothing. bootmgr works fine though)
I patched QEMU to have DBG2 in UEFI, and it does have kernel debug messages when running AArch64 full emulation, but nothing on KVM...

woachk, 2019-01-15:
@Googulator "if acpi=force is used on the command line, the kernel will attempt to use ACPI tables first, but fall back to DT if there are no ACPI tables present."

closed 2020-02-19 by andreiw