andreiw/RaspberryPiPkg #85: Two devices in the 3B+ IoT image's DSDT missing from the ACPI tables

Googulator, 2018-10-09:
The Pi 3 B+ IoT insider image contains the following 2 devices in its DSDT table:

` 
       Device (DSEN)
        {
            Name (_HID, "DSN3832")  // _HID: Hardware ID
            Name (_CID, "ACPIDSN3832")  // _CID: Compatible ID
            Name (_UID, Zero)  // _UID: Unique ID
            Method (_STA, 0, NotSerialized)  // _STA: Status
            {
                Return (0x0F)
            }
        }
`

`
        Device (FT54)
        {
            Name (_HID, "FTSC5406")  // _HID: Hardware ID
            Name (_CID, "FTSC5406")  // _CID: Compatible ID
            Name (_UID, Zero)  // _UID: Unique ID
            Name (_DEP, Package (0x01)  // _DEP: Dependencies
            {
                "\\_SB.RPIQ"
            })
            Method (_STA, 0, NotSerialized)  // _STA: Status
            {
                Return (0x0F)
            }

            Method (_CRS, 0, NotSerialized)  // _CRS: Current Resource Settings
            {
                Name (RBUF, ResourceTemplate ()
                {
                    Memory32Fixed (ReadOnly,
                        0x3E40A860,         // Address Base
                        0x00000040,         // Address Length
                        )
                })
                Return (RBUF)
            }
        }
`

No idea what these actually are. They aren't found in any form in our ACPI tables.

Googulator, 2018-10-09:
FTSC5406 is apparently the touchscreen digitizer part of the official Pi display.

Googulator, 2018-10-09:
DSN3832 is "BootVolumeSentinel Device", whatever that means

andreiw, 2018-10-09:
I could add them, sure. Don’t know what the sentinel device is about.

A

stefan2225, 2018-10-10:
It's probably something Windows 10 specific, only found it mentioned with Windows 10 and PCIE together on some kind of lists with drivers on google and everywhere else

driver1998, 2018-10-11:
Is these even used by Windows? The 3B image has these devices as well, but Windows doesn’t even have drivers for them.

发送自 Windows 10 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>应用

andreiw, 2018-10-11:
It could be related to an evolution of the crazy way the GPT partition table followed the boot FAT partition for MS-IoT. Or not.

It’s not immediately obvious what adding these adds, because we’re not talking about a hw device.

At the moment, not considering adding these anymore.

A

closed 2020-02-19 by andreiw