andreiw/RaspberryPiPkg #86: Installing unRAID on UEFI (Possible?)

moltenteam, 2018-10-21:


BenNottelling, 2018-10-21:
unRAID isn't available on ARM64 which is what the raspberry PI uses, but you may be able to find an ARM64 alternative

moltenteam, 2018-10-21:
Hello,
I was trying to install unRAID (unraid.net) onto Raspberry Pi 3 Model B with this UEFI. (unRAID is a Linux based OS) I normally imaged unRAID with unRAID USB Creator, enabled Allow UEFI Boot. When I booted to UEFI, it launched UEFI Interactive Shell. I typed in bootx64.efi (the unRAID UEFI bootloader) and it said Unsupported. Is even possible to install unRAID to RPi 3?

Thanks for any replies and sorry for my bad english.

moltenteam, 2018-10-21:
Thank you, I'll try it.

moltenteam, 2018-10-21:
OK, my plan is now to install qemu onto raspbian then emulate unRAID. Thank you for your help, BenNotteling.

closed 2018-10-21 by moltenteam

andreiw, 2018-10-23:
You're trying to run a 64-bit x86 bootloader (bootx64) instead of a 64-bit Arm one (bootaa64)

Your plan shouldn't be trying to emulate unRAID. That would be awful. You need to rebuild unRAID for Arm.

moltenteam, 2018-12-23:
Oh, thank you @andreiw