andreiw/RaspberryPiPkg #91: MX3 Air Keyboard Not working

JulyIGHOR, 2018-12-15:
I got this keyboard/mouse https://www.amazon.com/Keyboard-AMGUR-Wireless-Learning-Raspberry/dp/B0158AR8QK
It is working on any OS without drivers, but it is not working with Tiano Core at all.
Is there any way to fix it?

andreiw, 2018-12-16:
Hypothetically, yes, but unlikely that will happen soon- I don’t have this keyboard. Try plugging into a different port, or get a different keyboard.

A

andreiw, 2018-12-16:
> Thanks for the answer. I have tried to plugin to each of USB ports no luck.
> Actually I have no other keyboard and purchased this one with only purpose
> to work with Titan Core :)
> Can I help you somehow with it? Capture USB signals or anything else?

Short of getting the keyboard and playing around with it, not really.
Sorry. Not a great answer, but it is what it is. The good news is that this
is the second recent report I've heard about non-working keyboards.... so
literally the next USB keyboard you pull out of a dumpster will probably
work!

A

JulyIGHOR, 2018-12-16:
Ok, if some day you will change your mind, here is more info.

I think it is USB hub since it is Mouse+Keyboard.

Device detected as "Smart Control" with ID device: 0x1001 and  ID vendor: 0x1915  (Nordic Semiconductor ASA)
Looks like it is based on nRF24LU1
http://uglyduck.vajn.icu/PDF/wireless/Nordic/nRF24LU1_Development_Kit_UG.pdf

andreiw, 2019-10-20:
If you have the time, check the latest /upstream/ Pi 3, which has a few USB host driver improvs

closed 2019-10-20 by andreiw

JulyIghor, 2019-10-20:
> If you have the time, check the latest /upstream/ Pi 3, which has a few USB host driver improvs

I don't have this keyboard any more. But thanks. I'll try another one some time later :)