andreiw/RaspberryPiPkg #100: Blue Screen Of Death

manamster, 2019-01-07:
I now get a blue screen of death when attempting to boot that says 
"ATTEMPTED EXECUTE OF NO EXECUTE MEMORY"
This happens every time it gets to the "Just a moment screen" when booting.
Please help. All I want to do is run WOA.

thchi12, 2019-01-07:
Well it seems you are running 1809 or newer?
You need an earlier version of wpprecorder.sys like the one in 1803. Or you can change an earlier build like 1803.

BenNottelling, 2019-01-07:
I'm running 1809 and I never had to do that for it to work.

pbatard, 2019-01-07:
I've created a couple of 1809 installation.

I have __always have had to replace `Wpprecorder.sys` when running from SD.

I can also vouch that, even outside of trying to run Windows 10 1809 on RPi, BSODs just like the one describe happen as soon as you try to run 1809 (initial release or re-release) from USB on a regular PC (a.k.a. _Windows To Go_). 

This is a __well known issue__ at this stage (you can google for `DRIVER UNLOADED WITHOUT CANCELLING OPERATION` and Windows 10 1809 for instance), because Microsoft screwed up a bit more than removing the "My Documents" folder in their latest release, and the changes they applied to `Wpprecorder.sys` between 1803 and 1809 don't seem to have been tested properly...

BenNottelling, 2019-01-07:
I had to change the SD card controller in the UEFI for it to work, but I can show you that it is indeed 1809

pbatard, 2019-01-07:
As I said, the `Wpprecorder.sys` occurs when Windows detects that it is running in _Windows To Go_ like mode. It does not occur when it thinks it is running in regular mode. I am not saying that it is impossible to run Windows 10 1809 without replacing `Wpprecorder.sys` (as a matter of fact, I had it boot fine once without changing anything), but I can however guarantee that there is an issue if you don't replace this driver and Windows sees the drive it is running from as removable.

Without any alternate information from OP, we can only assume that they did not switch the SD card controller, and therefore are running into an issue that I am positive everybody will run to, with the default controller, when trying to run 1809 without replacing `Wpprecorder.sys`, especially as, like I mentioned before, this issue is __NOT__ limited to RPi, but will happen for any regular PC when you try to run 1809 from USB in _Windows To Go_ mode.

manamster, 2019-01-07:
by changing the controller it now boots up but when going through the setup I can't get past the account screen where there is the error "OOBELOCAL"

andreiw, 2019-01-14:
I've asked Mario to document the wpprecorder stuff discussed in this ticket. Since its not a UEFI issue, I am closing it away, even though it doesn't look like you've solved your problems. Use the MEGA thread. That's the only Windows thread I am going to leave around.

closed 2019-01-14 by andreiw