andreiw/RaspberryPiPkg #103: UEFI not booting to windows 10

Oky12, 2019-01-15:
hi ... I want to report a problem when testing the latest version of UEFI(Jan 14th, 2019).
the problem I found was that UEFI didn't boot into Windows 10.
but only shows the command shell after the splash screen. (see screenshot)
[image]

Oky12, 2019-01-15:
maybe this is the cause of my problem. when I open the boot manager. there are only 2 boot options. UEFI shell and UEFI misc devices. 

[image]
@andreiw 
where is another boot? (USB boot and uSD boot)

BenNottelling, 2019-01-15:
Why have you made this issue 3 times? If the first wrong, just edit it. Right now your spamming people's email boxes

Oky12, 2019-01-15:
@BenNottelling sorry about that. when making an issue. my network is slow. when I click submit new issue. I did not know that I made 3 similar issues

andreiw, 2019-01-16:
Okay, new Jan16th release will improve this.

UEFI will now *only* list devices plugged in.  You will also need to set the boot order the first time you copy a fresh UEFI build. Of course, the boot order will persist.

With USB plugged in:
[image]
With USB plugged in:
[image]

closed 2019-01-16 by andreiw