andreiw/RaspberryPiPkg #106: Enable SPI1 for AUXSPI driver to load properly

driver1998, 2019-01-15:
The current Windows bcmauxspi driver checks if SPI1 is enabled in the registers, if not, then the load fails, unless ForceEnable is set in the Windows registry. (Then the driver enables SPI1 and continue)

bcmauxspi.cpp 
```cpp

    ULONG enableMask;
    volatile BCM_AUXSPI_REGISTERS* registersPtr;
    switch (peripheralOffset) {
    case FIELD_OFFSET(BCM_AUX_REGISTERS, Spi1):
        enableMask = 0x2;   // Spi1Enable
        registersPtr = &thisPtr->auxRegistersPtr->Spi1;
        break;
    case FIELD_OFFSET(BCM_AUX_REGISTERS, Spi2):
        enableMask = 0x4;   // Spi2Enable
        registersPtr = &thisPtr->auxRegistersPtr->Spi2;
        break;
    default:
        NT_ASSERT(!"peripheralOffset should have been validated above");
        return STATUS_INTERNAL_ERROR;
    }

    //
    // Ensure device is enabled. This is a shared register for all devices
    // on the AUX peripheral, so we cannot safely modify it without
    // synchronizing with all the other AUX devices. If the peripheral is not
    // enabled, fail the load of the driver.
    //
    BCM_AUX_ENABLES_REG enablesReg =
        {READ_REGISTER_NOFENCE_ULONG(&thisPtr->auxRegistersPtr->Enables)};

    if (!(enablesReg.AsUlong & enableMask)) {
        NTSTATUS status = queryForceEnableSetting(WdfDeviceGetDriver(WdfDevice));
        if (!NT_SUCCESS(status)) {
            AUXSPI_LOG_ERROR(
                "The device is not enabled. The device must be enabled in the "
                "AUX_ENABLES register prior to driver load. "
                "(enablesReg.AsUlong = 0x%x, enableMask = 0x%x)",
                enablesReg.AsUlong,
                enableMask);
            WdfDeviceSetFailed(WdfDevice, WdfDeviceFailedNoRestart);
            return STATUS_DEVICE_HARDWARE_ERROR;
        }
        //.....
```

And it seems currently SPI1 is not enabled by default, requiring ForceEnable=1 in the registry.

driver1998, 2019-01-15:
I can patch out the check for sure, but it will be better to do the job on UEFI side.

andreiw, 2019-01-15:
Ok, will do.

driver1998, 2019-07-04:
I ended up added the ForceEnable DWORD in the bcmauxspi INF file anyway, so it should be fine to close this.

closed 2019-07-04 by driver1998