andreiw/RaspberryPiPkg #110: Full support for overclocking

BenNottelling, 2019-02-13:
I've looked into overclocking the PI in  the past, but couldn't get much results (short of 1450 on the arm freq, up from 1400)
My question is, does the firmware support it properly, can we set things such as ram speed, voltage, GPU speed, ect? I'm assuming that I can't get a higher OC because it needs more voltage, but changing it seems to do nothing.
Thanks!

mariobalanica, 2019-02-13:
The clock rates can be dynamically changed through the mailbox interface.

Setting the core voltage on SoCs newer than BCM2835 isn't possible due to the AVS implementation in the RPi firmware, so this has to be manually changed from the config.txt file.

I may submit a pull request that adds support for setting arbitrary clock rates this week.

BenNottelling, 2019-02-13:
I was using the config.txt file
Inside the boot partition from using WOR

mariobalanica, 2019-02-13:
Raspberry Pi 3 B+ is running quite close to the maximum clock rate already.

andreiw, 2019-02-13:
You can basically use config.txt for all your overclocking needs, as before.

A

andreiw, 2019-02-13:
Sure, looking forward to your pull request. Note, for the CPU I was not able to set any intermediate freqs between 600 and max. It was always one or the other.

A

mariobalanica, 2019-02-13:
Yes, when setting the ARM clock rate to a value higher than 600 MHz the turbo mode kicks in by default.
In order to avoid that, the "skip setting turbo" value should always be 1.

BenNottelling, 2019-02-13:
[image]
@mariobalanica So are you saying that because I didn't set `skip setting turbo` this isn't running at 1.46Ghz?

andreiw, 2019-02-14:
No, I think what he was saying is that setting it to 800Mhz or 1337Mhz would not be possible without skipping that turbo setting.

A

BenNottelling, 2019-02-14:
Gotcha

mariobalanica, 2019-02-14:
I'm talking about the mailbox stuff.
That behaviour is documented here: https://github.com/raspberrypi/firmware/wiki/Mailbox-property-interface#set-clock-rate

mariobalanica, 2019-02-21:
https://github.com/andreiw/RaspberryPiPkg/pull/117

andreiw, 2019-02-21:
Merged

andreiw, 2019-02-21:
Merged

closed 2019-02-21 by andreiw