andreiw/RaspberryPiPkg #114: WoA won't boot

himike123, 2019-02-16:
Im having a issue with WoA not booting  it's stays on the screen were it says "Exiting UEFI and booting OS kernel" in trying to boot from USB since my desktop does not have a SD card reader, I would try to go in the bootloader and change boot option but it won't show my USB all it shows is "port 1 port 2 port 3 port 4" any help but be grateful

andreiw, 2019-02-17:
You’re using a really old uefi firmware build. For sure. Use the latest.

What windows build are you booting? Did you change hypdxe settings to boot in EL1 instead of EL2?

A

himike123, 2019-02-17:
Thanks for the fast reply, I'll try changing the hypdxe and install the latest UEFI, also may you please link the latest UEFI software for me please?

andreiw, 2019-02-17:
https://github.com/andreiw/RaspberryPiPkg/tree/master/Binary/prebuilt/2019Jan16-GCC5/RELEASE

closed 2019-02-17 by andreiw