andreiw/RaspberryPiPkg #116: Not able to go back to broadcom driver in boot

zopit, 2019-02-20:
I installed everthing correct but after the complete install of windows not able to go into the boot menu because time is like milisecond and it start the window process and now the its slow because of the driver pls help ??

BenNottelling, 2019-02-20:
Just spam the escape key..

zopit, 2019-02-20:
Already tried the every comination not able to get to the boot menu any other hint ?

BenNottelling, 2019-02-20:
There is no combination, it's a just the escape key
I've done this all the time, so I'm not sure what else might be going on

mariobalanica, 2019-02-21:
Type "exit" at the EFI shell.

zopit, 2019-02-22:
I think i hve to install comletely one more time and increase the time of boot menu from 5 to 10 sec :unamused:

andreiw, 2019-02-28:
You can try serial port as well. What I do is spam the esc key as soon as ai apply power (do not wait for video to come up)

Alternatively, just place a fresh rpi_rfi.fd and try configuration again.

closed 2019-02-28 by andreiw