andreiw/RaspberryPiPkg #125: WIN10 (ARM) reboot after BIOS setup

joke1014, 2019-04-02:
WIN10 (ARM) version has been saved to SD card

The raspberry pie is energized after inserting keyboard, mouse and screen

Press ESC to enter BIOS and set it as pictures

The problem in the last photo occurs when the keyboard is removed and the mouse is restarted.

How can I solve this problem?
[image]
[image]
[image]
[image]
[image]
[image]
[image]

BenNottelling, 2019-04-02:
You need to change your boot order

joke1014, 2019-04-02:
> 您需要更改引導順序

Can you tell me exactly how to do it?

joke1014, 2019-04-02:
> You need to change your boot order

Can you tell me exactly how to do it?

joke1014, 2019-04-02:
> You need to change your boot order

Can you tell me exactly how to do it?
Because both BOOT EL1 and EL2 have tried it.

laurinneff, 2019-08-13:
I don't have this on my SD yet, but I'm guessing it's Boot Manager in the main menu

andreiw, 2019-10-20:
Boot manager -> boot order -> make the usb key with woa first

On upstreamed edk2-platforms this issue no longer exists

closed 2019-10-20 by andreiw

mullcom, 2019-10-20:
How do I get this bios on raspberry?