andreiw/RaspberryPiPkg #126: Raspberry pi 3 b+ BIOS W10

albapv, 2019-04-02:
I have installed the version of Windows 1089 10 arm64, following all the steps through WoR Installer, LAN Drivers for r 3 b+, Drivers_RPi3_20Feb19_USBBoot and RPI_EFI.fd.

The procedure has finished correctly, but when you insert the micro SD and turn on the Raspberry, nothing appears on the screen, not even an image connection arrives.
I've tried to install it several times and I can't make it work.

albapv, 2019-04-03:
I add data: I use the official raspberry power adapter and a hdmi to vga display adapter

andreiw, 2019-10-20:
No idea. A debug build if uefi and a serial adapter could shed light.

Did the boot part get wiped somehow? Not fat32? Unsupported sd card?

closed 2019-10-20 by andreiw