andreiw/RaspberryPiPkg #127: [Feature Request] add support to MultiBoot With GRUB2

Oky12, 2019-04-04:
I have a project to create a multi OS on raspberry PI with GRUB2 as a tool to choose boot OS. but unfortunately UEFI does not support GRUB2.

I tried it on a flashdisk which was formatted as fat32 and installed GRUB2 version 2.02 UEFI 64bit efi

The OS that I made multi boot is Windows 10 Arm64 and UBUNTU or other debian

if my request cannot be applied. I will try to make a multiboot with the Windows Boot Manager

driver1998, 2019-04-04:
this doesn't need special support in UEFI, you can let grub2 to chainload bootmgrfw.efi, or add both grub2 and bootmgr to the UEFI boot menu.

Oky12, 2019-04-04:
so. I have to add GRUB2 to the Windows Boot Manager?

I want UEFI to boot into GRUB2 and then in GRUB2 we choose to boot the OS

andreiw, 2019-10-20:
Should work out of the box (eg Susr uses Grub2)

closed 2019-10-20 by andreiw