andreiw/RaspberryPiPkg #129: How to use PL011 UART on GPIO14&GPIO15

GitHub-Hugl, 2019-06-21:
I have installed full win10 pro on my raspberrypi 3b+.It works! But i want to use GPIO14 and GPIO15 for hard serial communication. I can send data via COM1,and my own io equipment also can receive the data , but raspberrypi 3b+ can not receive feedback from my own io equipment. 

 I have an idea ,if we can change  UEFI fILE on  the Raspberry Pi 3+ to disable bluetooth ?

GitHub-Hugl, 2019-06-21:
Some documents shows that my question was becaused by raspberrypi 3b+ configured GPIO14 &GPIO15 to mini-uart,    and  PL011 UART to bluetooth !

mariobalanica, 2019-06-22:
PL011 UART does not work, but you can use the mini UART.

GitHub-Hugl, 2019-06-22:
I can send data to my io equipment by  mini uart successfully  ,but I can't receive the data  from my io via mini uart.  They both use 3.3v power发自我的华为手机-------- 原始邮件 --------主题：Re: [andreiw/RaspberryPiPkg] How to use PL011 UART on GPIO14&GPIO15 (#129)发件人：Mario Bălănică 收件人：andreiw/RaspberryPiPkg 抄送：GitHub-Hugl <852266205@qq.com>,Author PL011 UART does not work, but you can use the mini UART.

mariobalanica, 2019-06-22:
Short GPIO14 and GPIO15 together, open a serial terminal (e.g. PuTTY) and start sending some keys. If the data gets echoed back then everything is OK on the RPi side.

andreiw, 2020-02-19:
This repo is deprecated and I don't check these conversations.

closed 2020-02-19 by andreiw