andreiw/RaspberryPiPkg #132: Replace encoded MsftFunctionConfigs with their standard equivalent, PinFunction

Googulator, 2019-06-26:
The MsftFunctionConfig macro got standardized in ACPI 6.2 as PinFunction, and although this isn't documented anywhere, recent builds of Windows 10 support the standard version (at least 18362 does - not sure when it got implemented).
Switching to PinFunction would be helpful for letting Linux boot via ACPI.

driver1998, 2019-06-26:
If iasl supports it (it should but I haven't checked), that also means we can get rid of VenderLong and have a more readable code.

Googulator, 2019-07-01:
Yes, it's supported in recent versions of iasl.

andreiw, 2019-07-01:
(tested) pull requests welcome.

driver1998, 2019-07-01:
Translated all the MsftFunctionConfigs to PinFunction (they are basically the same), and Windows 10 b18362 seems happy with that: 
[image]

I don't have any SPI device to test so far, but I think it will work fine as well.

pbatard, 2019-07-02:
See my [comment on PR #134](https://github.com/andreiw/RaspberryPiPkg/pull/134#issuecomment-507611907).

driver1998, 2019-07-02:
So PinFunction is not supported in v1809 (17763), with the same behavior like https://github.com/andreiw/RaspberryPiPkg/issues/96.

pbatard, 2019-07-02:
My understanding is that this has nothing to do with the Windows version being used, but everything to do with whether the `iasl` compiler, which is executed to parse the `.asl` when compiling the UEFI firmware, and which could be executed from any platform (but would mostly be from Linux) potentially being "too old" to be understood by the compiler.

In other words, in `PinFunction()` was introduced to the ACPI specs after 2016, then any ACPI pre-2106 compiler that attempts to compile a file that invokes it will produce an error, which is what I am seeing when trying to compile `DSDT.asl` on Debian 9.9 (latest), because the native `iasl` it provides, which comes from the acpica-tools package, dates from 2016...

driver1998, 2019-07-02:
@pbatard This is a recent addition to ACPI spec (in ACPI 6.2), so it is not surprised to see it not supported in old Windows versions.

I use the same DSDT compiled in iasl 20180105, and test it in both 17763 and 18362.

pbatard, 2019-07-02:
> so it is not surprised to see it not supported in old Windows versions.

Again, this is __not__ a matter of Windows not supporting it. It's a matter of the __Linux__ ACPI compiler not supporting it.

AFAIK, all Windows sees are the compiled binary blobs, and it doesn't care whether they were produced by an ACPI compiler that used `PinFunction()` or `MsftFunctionConfig()` or `VendorLong()` since the end result will be exactly the same.

The only issue here is whether the compiler understands any of the above functions to create the binary blob, and the current default Linux ACPI compiler on Debian only knows about `VendorLong()` because it was created in 2016, before `PinFunction()` was introduced.

driver1998, 2019-07-02:
I am using iasl 20180105, which has no problems with PinFunction, it compiles the file just fine.

The output is recognized in 18362, but **NOT** 17763.

pbatard, 2019-07-02:
> I am using iasl 20180105

And as I pointed out, the default `iasl` from Debian 9.9, which is the most up to date Debian, dates from 2016, and doesn't have `PinFunction`.

So, once again, I'll leave it to Andrei to decide whether he wants to require people to have to manually upgrade their `iasl`, since it is likely that the default version provided by their platform will be too old to understand `PinFunction`, or wait until platforms like Debian have upgraded their default `iasl` package to include a version that supports `PinFunction` (such as 20180105).

Please bear in mind that not everybody wants to have to __manually__ update toolchain components on their system, because it can create maintenance headaches. My guess is that a lot of people expect that compilation should work with the default system toolchain (especially if their system is up to date) and that, because there is always a lot of latency between what a toolchain provides and latest compilers and tools, we therefore may have to wait for a while before we see a native `iasl` on Debian that supports `PinFunction`...

driver1998, 2019-07-02:
Yes but that is a completely different issue...
BTW, the next major release of Debian, Buster, comes with iasl 20181213, and is planned to release by July 6th. Although it is also annoying to require everybody to have nothing but the very latest release...

andreiw, 2019-07-02:
Considering there is a difference in output AML, and subsequent difference in supported builds, I’d hide this under a macro (and preprocess the ASL prior to running it through iasl), with the default implementation being the old (and compatible with old iasl)

A

driver1998, 2019-07-04:
As I mentioned in https://github.com/andreiw/RaspberryPiPkg/pull/134#issuecomment-507813600:

Now you can specify -DACPI_PINFUNCTION=1 in the build commandline to use PinFunction, by default it still uses MsftFunctionConfig.

andreiw, 2019-07-05:
> As I mentioned in #134 (comment):

Thanks for your contribution - it’s merged.

A

closed 2020-02-19 by andreiw