andreiw/RaspberryPiPkg #133: Ability to load extra SSDTs from file, to support HATs

Googulator, 2019-07-01:
In the Linux world, HATs are supported using device tree overlays loaded either manually through config.txt, or automatically from an onboard EEPROM.
The analogous functionality to device tree overlays in ACPI is SSDTs. It would be great if UEFI provided a way of loading a custom SSDT from file, or even from the HAT EEPROM.

andreiw, 2019-07-01:
Sure, but, what's ACPI Linux on the Pi 3? It's possible, but no one is doing it (that I am aware of). And the value proposition of that is pretty low (esp. because of HAT EEPROM, and because Pi 3 is so non-standard... Pi 4 would be better suited to ACPI).

Assuming you booted in ACPI mode, you'd be better off converting the HAT EEPROM DTB overlay on the fly to a SSDT with _DSD

I'll keep this open for others to see and comment on (though the upstream edk2 mailing list is likely a better place for a call-for-action)

Googulator, 2019-07-02:
I was primarily thinking of Windows here, not Linux. Linux can continue to support HATs using the existing DTB overlay mechanism, but Windows has no chance of ever supporting DTB, so an ACPI analog is needed.

driver1998, 2019-07-02:
For Windows, you can take a look at how Microsoft do that in IoT Core.
This is the sample driver for a SPI LCD display:

https://www.hackster.io/windows-iot/windows-10-iot-core-for-adafruit-spi-touchscreen-bb3795

andreiw, 2019-07-02:
Oh ok, that makes sense. It should be pretty trivial to find and load SSDTs from storage.

A

andreiw, 2020-02-19:
This repo is deprecated and I don't check these conversations.

closed 2020-02-19 by andreiw