andreiw/RaspberryPiPkg #136: [error] Can not boot from 4.19 on Rpi3b+

aiminickwong, 2019-07-02:
When I upgrade Kernel from 4.14 to 4.19, I got the black screen with Rpi3b+, but with kernel at version 4.14 is OK.

andreiw, 2019-07-06:
A serial log would help.

malignus, 2019-09-23:
Hi,
same problem here. Captured a serial log.
[boot.log](https://github.com/andreiw/RaspberryPiPkg/files/3641103/boot.log)

pbatard, 2019-09-23:
@malignus,

Do you see the same issue with both the DEBUG an RELEASE firmware or just DEBUG?
Also can you try the firmware binary from https://github.com/pbatard/RPi3/releases and report?

This is a slightly different firmware binary from Andrei's, that is generated from the now official sources for the Raspberry Pi 3 UEFI platform we have in [ed2-platforms](https://github.com/tianocore/edk2-platforms/tree/master/Platform/RaspberryPi/RPi3), and that also contains a handful of fixes that aren't in Andrei's tree, such as for instance some automated patching of the Device Tree at firmware boot time so that USB keyboards properly work with Debian [tianocore/edk2-platforms@5f003136c2bf86620c0caaef0ad146b5e7fd2f6b] as well as other stuff. Oh and this firmware also embeds the latest device tree for the Pi 3 that were updated by the Pi Foundation on the Pi 4 release, which can fix the ability to use the serial port in Linux.

I have no issue running kernel 4.19 using the firmware linked above, on a Debian 10.x environment that I installed by following [this guide](https://pete.akeo.ie/2019/07/installing-debian-arm64-on-raspberry-pi.html).

pbatard, 2019-09-23:
Interesting... I have just tested the [2019.02.18 firmware release from this repo](https://github.com/andreiw/RaspberryPiPkg/tree/master/Binary/prebuilt/2019Feb18-GCC5/RELEASE) (latest) with my Debian 10 installation (that is running Debian kernel 4.19.0-6-arm64) and found that it froze on boot as reported above.

But the [2019.01.16 one](https://github.com/andreiw/RaspberryPiPkg/tree/master/Binary/prebuilt/2019Jan16-GCC5/RELEASE) seems to be okay...

malignus, 2019-09-23:
Hi,

tested the firmware from here: https://github.com/pbatard/RPi3/releases and it worked.

andreiw, 2019-10-20:
Closing because while it’s a real issue, it won’t be investigated in this repo, now that this UEFI has been upstreamed to official Tiano trees

closed 2019-10-20 by andreiw