andreiw/RaspberryPiPkg #137: Drivers

HeyGuysWassup, 2019-07-05:
Any update on the drivers as if now 
I may have found a wifi driver: https://www.driver-wireless.com/broadcom-bcm43xx-driver-win7810.html
Anyway to convert x86 or x64 driver to ARM64 with ease? Can't you use windows iot drivers because isn't ARM32 drivers backwards-compatible with ARM64?

HeyGuysWassup, 2019-07-05:
Any update also if when element14 will release uefi firmware for raspberry pi 4? Any ETA for that or is there any ETA for windows 10 on raspberry pi 4.

TheBobPony, 2019-07-05:
> Any update on the drivers as if now
> I may have found a wifi driver: https://www.driver-wireless.com/broadcom-bcm43xx-driver-win7810.html
> Anyway to convert x86 or x64 driver to ARM64 with ease? Can't you use windows iot drivers because isn't ARM32 drivers backwards-compatible with ARM64?

If there's the source code for it then it will be possible to convert them into ARM64 drivers, and ARM32 drivers don't work on ARM64 Windows 10, same thing for trying to install 32-bit drivers on 64-bit Windows.

HeyGuysWassup, 2019-07-05:
Sorry for the delay. Thanks for the response! On "https://discourse.pi64.win/t/status-of-the-arm64-device-drivers-as-of-12-feb-2019/38", can the drivers please be updated for what's working and what's not? Also, Any update also if when element14 will release uefi firmware for raspberry pi 4? Any ETA for that or is there any ETA for windows 10 on raspberry pi 4.

andreiw, 2019-07-06:
Element114 is working on UEFI for Pi4? If they are, then you're asking in the wrong place. If you're asking if I'm working on Pi 4 UEFI support, then the short answer is - not at the moment.

closed 2019-07-06 by andreiw

HeyGuysWassup, 2019-07-06:
Oh ok