andreiw/RaspberryPiPkg #138: WD Pi Drive shuts down during boot. (USB Boot problems)

MikeDev101, 2019-07-23:
Let's just start of by saying I'm a noob with this project, and GitHub in general. I know that there were a few posts similar to usb booting problems, and also how molasses slow the SD interface is. I tried copying the now ready "Windows On Raspberry Pi" Micro SD card to my 250GB Western Digital Pi Drive. It loaded the UEFI boot manager fine, but after it begins loading Windows it shuts down the USB interface and also shuts off the HDD, resulting in a BSoD "INACCESSIBLE_BOOT_DEVICE". Any suggestions?

mariobalanica, 2019-07-23:
You need a powered USB hub.
Keep in mind that Windows is way slower than Linux on the Raspberry Pi.

MikeDev101, 2019-07-23:
But I was already doing that? It shuts down the USB interface, but my drive still has power. It assumes it got unmounted, and goes into some sort of "sleep mode".

mariobalanica, 2019-07-23:
What tool have you used to install Windows on the SD card?

andreiw, 2020-02-19:
This repo is deprecated and I don't check these conversations.

closed 2020-02-19 by andreiw