andreiw/RaspberryPiPkg #140: Network Booting UEFI for Network Booting Kernel

gordan-bobic, 2019-10-13:
Thanks for an awesomely useful project to give the Pi a standards compliant boot loader.

I have UEFI booting via DHCP+TFTP without a local storage device on the Pi. This works great, but I would like to, from there, TFTP download and boot the dtb/kernel/inirtd. This leads me to two issues:

1) There appears to be no TFTP command in the shell. Is there a way to gain that?

2) Once I can get past 1), I still have the issue of instructing the TFTP booted Tianocore to "do stuff", either
2.1) via the configuration menu which has no local storage to save the NVRAM to (does it?)
2.2) via sending a custom startup.nsh via TFTP

How do I go about adding TFTP (or HTTP, which I am lead to believe is also an option) option to this Tianocore build, and how do I go about pushing out a custom startup.nsh script that makes use of TFTP to fetch and boot a dtb/kernel/initrd?

andreiw, 2019-10-20:
Yes. The tftp is just a shell built in - https://github.com/freebsd/uefi-edk2/tree/master/ShellPkg/Library/UefiShellTftpCommandLib. Grep for the inf file in an edk2 tree to see how to include it in the DSC.

Good news is that you can just build your own shell and load it without building the fw.

“NVRAM” support is there but requires local storage.

To push startup.nsh you’ll probably want to look at FvSimpleFilesystem.

Note that this tree is officially dead - its been upstreamed into edk2-platforms. Don’t use this tree. It’s been superseded.

Note that the upstream tree does not boot into UEFI by default any longer. Sigh.

closed 2019-10-20 by andreiw

pbatard, 2019-10-20:
Note that this issue is also a duplicate of pbatard/RPi3#3.

The latest builds I carry of the upstreamed Raspberry Pi firmware ([here](https://github.com/pbatard/RPi3/releases)) now include `tftp` in the Shell, as per tianocore/edk2-platforms@f8c5d83e0042e5f4c42d5e9fdd88a92e449b156c.